gpiozero/gpiozero #464: Device = 1 on SPI not working

Grumpy-Mike, 2016-09-23:
using MCP3008(channel=0, device=1) or MCP3008(channel=0, device=0) results in the same pin being used for chip select, namely GPIO8

Even if this did work, how would you have two MCP3008s on the SPI bus one using GPIO8 and the other using GPIO7 as the chip address?

Python 2 - gpio zero versions 1.0.0 and 1.3.1

lurch, 2016-09-23:
Looking at what https://github.com/RPi-Distro/python-gpiozero/blob/master/gpiozero/spi.py#L26 is doing, I wonder if this might be a bug in spidev?

@waveform80 on a slight side-note (given that this is the first time I've properly looked at the SPI code), I think it would have made more sense for SPISoftwareInterface to descend from CompositeDevice (since an SPI interface is composed of multiple pins) rather than from OutputDevice?

@Grumpy-Mike Isn't [this](https://en.wikipedia.org/wiki/Serial_Peripheral_Interface_Bus#Independent_slave_configuration) how it's supposed to work?

joan2937, 2016-09-23:
It appears to work correctly for me.
```py
#!/usr/bin/env python

import gpiozero

d1=gpiozero.MCP3008(channel=0, device=0)
d2=gpiozero.MCP3008(channel=0, device=1)

v1 = d1.value
v2 = d2.value

print(v1, v2)

d1.close()
d2.close()
```
CE0(8)/CE1(7) are used for d1/d2.

I'm using the version of gpiozero bundled with jessie (1.3.1).

Grumpy-Mike, 2016-09-23:
True that does. What does not work is
`#!/usr/bin/env python
from gpiozero import MCP3008

MCP3008(channel=0,device=1)

print"Twin A/D board test"
while True:
     a2d0 = MCP3008(0)  
     print" ",a2d0.raw_value`
Which makes this a problem with the documentation which suggests that it would work.

lurch, 2016-09-23:
I'm not sure I follow? By doing `MCP3008(channel=0,device=1)` you're creating an ADC on CS1, which instantly gets garbage-collected (because you're not storing the returned object anywhere). And then when you later do `a2d0 = MCP3008(0)` you're creating an ADC on CS0, because that's the default when no `device` is explicitly specified.

I wonder if this is more what you're trying to do?
```python
#!/usr/bin/env python
from gpiozero import MCP3008

a1d0 = MCP3008(channel=0,device=0) # the same as: a1d0 = MCP3008(0)
a2d0 = MCP3008(channel=0,device=1)

print"Twin A/D board test"
while True:
     print" ",a2d0.raw_value
```

Grumpy-Mike, 2016-09-23:
No that is the exact code, and I was following the documentation that mentioned nothing of the 

> which instantly gets garbage-collected (because you're not storing the returned object anywhere).
That's why I said the documentation needs to tell people how to use this call.
@joan2937 - how do you change the channel of the A/D you are reading with that code?

joan2937, 2016-09-23:
This works okay and uses the second channel select (CE1).
```py
#!/usr/bin/env python

from gpiozero import MCP3008

d = MCP3008(channel=0, device=1)

print(d.value)

d.close()
```
@Grumpy-Mike I'm not sure you can on the fly with gpiozero.  The following seems to work.
```py
#!/usr/bin/env python

from gpiozero import MCP3008

d0_0 = MCP3008(channel=0, device=0)
d0_1 = MCP3008(channel=1, device=0)

print(d0_0.value, d0_1.value)

d0_0.close()
d0_1.close()
```

lurch, 2016-09-23:
I guess you're referring to the "the following constructors are all equiavlent" part of https://gpiozero.readthedocs.io/en/latest/api_spi.html#spi-keyword-args ?  Yeah, that's possibly misleading, with the result of the constructor not being stored anywhere.

> how do you change the channel of the A/D you are reading

With the current code (until #429 gets worked on) you need to create a separate object for each channel of the ADC. So going back to my earlier example again:
```python
#!/usr/bin/env python
from gpiozero import MCP3008

# eight channels on the MCP3008 connected to CS0
a1d0 = MCP3008(channel=0,device=0)
a1d1 = MCP3008(channel=1,device=0)
a1d2 = MCP3008(channel=2,device=0)
a1d3 = MCP3008(channel=3,device=0)
a1d4 = MCP3008(channel=4,device=0)
a1d5 = MCP3008(channel=5,device=0)
a1d6 = MCP3008(channel=6,device=0)
a1d7 = MCP3008(channel=7,device=0)

# eight channels on the MCP3008 connected to CS1
a2d0 = MCP3008(channel=0,device=1)
a2d1 = MCP3008(channel=1,device=1)
a2d2 = MCP3008(channel=2,device=1)
a2d3 = MCP3008(channel=3,device=1)
a2d4 = MCP3008(channel=4,device=1)
a2d5 = MCP3008(channel=5,device=1)
a2d6 = MCP3008(channel=6,device=1)
a2d7 = MCP3008(channel=7,device=1)

print"Twin A/D board test"
while True:
     print "ADC2 channel 0 ",a2d0.raw_value
     print "ADC1 channel 4 ",a1d4.raw_value
```

Grumpy-Mike, 2016-09-23:
> With the current code (until #429 gets worked on) you need to create a separate object for each channel of the ADC. 

Wow guys that is so lame. I guess I will abandon gpio zero at least until this is sorted out.
Thanks for clarifying things.

joan2937, 2016-09-23:
@Grumpy-Mike That's a tad unfair to the gpiozero developers.  You have to consider their target audience.  For what you want you would be better off using something like pigpio directly.

lurch, 2016-09-23:
...of course there's no reason you couldn't make it neater by using lists of objects:
```python
#!/usr/bin/env python
from gpiozero import MCP3008

# eight channels on the MCP3008 connected to CS0
adc1 = []

# eight channels on the MCP3008 connected to CS1
adc2 = []

for c in range(8):
    adc1.append(MCP3008(channel=c, device=0))
    adc2.append(MCP3008(channel=c, device=1))

print "Twin A/D board test"
while True:
     print "ADC2 channel 0 ",adc2[0].raw_value
     print "ADC1 channel 4 ",adc1[4].raw_value
```

Grumpy-Mike, 2016-09-23:
> You have to consider their target audience.

I am and that is why I said the documentation needs work to give actually valid examples that can be built on, not examples that only work by chance.

@lurch - Thanks for that. I had actually considered a list of them but I think wiringPi is the best bet for this project I am working on at the moment.

closed 2016-09-23 by Grumpy-Mike

lurch, 2016-09-23:
Yeah, like a lot of OpenSource projects, the documentation is something that could do with more work.
But to be fair, the beginners that GpioZero is aimed at are unlikely to be using *two* eight-channel ADCs! The https://gpiozero.readthedocs.io/en/latest/recipes.html page is where we expect most beginners to start.

waveform80, 2016-09-23:
> Wow guys that is so lame. I guess I will abandon gpio zero at least until this is sorted out.
> Thanks for clarifying things.

To be perfectly frank, it's comments like this that make me think "you're more than welcome to go elsewhere" (you can imagine slightly more colourful language here if you wish).

You might've offered to put some code where your mouth is ... but somehow I don't see that happening given you didn't understand Python object lifetimes. You might've offered to improve the documentation with your new-found knowledge, courtesy of the other people here. At the very least, the *bare minimum* you could've managed would be to thank the others for examining the problem, explaining it, and admitting that our docs aren't perfect (that's my fault, don't blame them for that).

But no, you'd rather fling mud and leave in a huff. Good riddance.

Grumpy-Mike, 2016-09-23:
> You might've offered to put some code where your mouth is ... but somehow I don't see that happening given you didn't understand Python object lifetimes. 

True I did not understand Python object lifetimes, so how do you expect a beginner to understand them? As to code documentation this is something that must be done by those who do understand Python objects. As to contributing I do have a thread going on the Pi forum and I am about to update that with elements of the answers that I found here.

> But no, you'd rather fling mud and leave in a huff. Good riddance.

Hey all I said was it was lame, if you want to see mud you have seen nothing yet. This is a typical "Linux experts" response.  Full of "what a weenier for not understanding that" coupled with "how dare you criticize us". First remove the beam from your own eye.

bennuttall, 2016-09-23:
I think this is rather unfair. The example you gave just didn't make sense. Admittedly the example you found in the docs could be misleading (it's not a demo of how you use the class, it's making a point about how the init params can be used), but those are the API docs, not the starter guide. For examples, see how the `MCP3008` is used in the [recipes page](http://gpiozero.readthedocs.io/en/v1.3.1/recipes.html#potentiometer).

Beginners usually start with LEDs and buttons, following simple examples and they will easily pick up the "format" of a gpiozero script: you create connections to devices e.g. `led = LED(pin)` and `button = Button(pin)`, and you interact with them based on the name you gave them. It's as simple as possible, without limiting general purpose use.

We've spent a *lot* of time and effort getting the API right. A lot of thought has gone into its consistency and ease of use for beginners. We're open to constructive criticism but you haven't given us any. You tried to do something the wrong way, and when we showed you how to do it right, you said you'd rather use another library. That's fine, but I don't think you've given it a fair trial. The basic premise of the library is that rather than talking of pins, we talk of devices. The same applies to ADCs: rather than talking of ADC channels, we talk about potentiometers and sensors:

```python
pot = MCP3008(0)
```

If you have multiple potentiometers, you can use regular Python shorthand to create a list of pots:

```python
pots = [MCP3008(c) for c in range(8)]
```

Though this method is available for multiple LEDs, we also provide a [LEDBoard](http://gpiozero.readthedocs.io/en/v1.3.1/api_boards.html#ledboard) class:

```python
leds = LEDBoard(2, 3, 4, 5, 6)
```

and we've recently been discussing doing the same for ADCs, which might make sense for what you're trying to do. See #429. If you have any suggestions, please comment. I'd advise you brush up on the library style by reading through the recipes page.

GPIO Zero was designed to be easier for beginners, children and teachers to use, by abstracting the logic of what's going on with the pins away, and letting them use devices, making their code explain what the components do, not how they do it, the same way any modern library would (think [requests](http://docs.python-requests.org/en/master/)). Choosing a pin based library over a trivial detail means putting yourself or others through unnecessary pain. The gains just don't add up.

waveform80, 2016-09-24:
> Hey all I said was it was lame, if you want to see mud you have seen nothing yet. This is a typical "Linux experts" response. Full of "what a weenier for not understanding that" coupled with "how dare you criticize us". First remove the beam from your own eye. 

No, I think I'll defuse this by offering a begrudging apology. My response was written after receiving an e-mail from another contributor which I apparently misinterpreted as "I've just spent a while dealing with someone thoroughly unpleasant, how about some backup?". I skimmed the thread, noted you'd simply complained without offering anything constructive, assumed your "thanks" at the end was entirely sarcastic, and decided to deliver a swift kicking before dashing off to pick up my daughter from nursery. So, my apologies for any over-reaction.

As for the rest of what I was going to say, I think Ben's covered it well enough already above so I'll just duck out for the remainder of this, and get back to wrangling with code rather than people.