gpiozero/gpiozero #466: Button "is_active()" crash !!!

debugasm, 2016-09-25:
Hi, I have used for first time gpiozero with this code on Raspberry Pi 3 :
```
import sys

from gpiozero import Button

input_hw_pin = Button(6, pull_up=False)

try:
    if input_hw_pin.is_active():
        print "Pin High"
    else:
        print "Pin Low"
except:
    print "Unexpected error:", sys.exc_info()[0]
```
But program carsh with:

exception: Unexpected error: <type 'exceptions.TypeError'>

Has anyone ever had a problem like that ?

debugasm

lurch, 2016-09-25:
`is_active` is a [property](http://gpiozero.readthedocs.io/en/v1.3.1/api_generic.html#gpiozero.Device.is_active), not a method, so you need to change `if input_hw_pin.is_active():` to `if input_hw_pin.is_active:` (i.e. remove the parentheses).

debugasm, 2016-09-26:
You are definitely right. I'm not an expert in Python, I am confused on watching this code:
```python
from gpiozero import LED
from time import sleep

led = LED(17)

while True:
    led.on()
    sleep(1)
    led.off()
    sleep(1)
```
and I imagined that "is_active" also had a method and not a property.

Thanks very much

closed 2016-09-26 by debugasm

bennuttall, 2016-09-26:
The error you should have got is:

```
TypeError: 'bool' object is not callable
```

Because `is_active` is either True or False, and you're trying to call it.