gpiozero/gpiozero #467: Pin numbering

waveform80, 2016-09-25:
Following on from #459 ... here's the contentious change I've currently got in the patch: pin numbering.

Yeah, that old chestnut. Funny story...

So, @bennuttall and I are at PyConUK and we bump into @gbaman and get chatting about GPIO Zero. Now, it turns out that @gbaman might be *almost* as stubborn as I am, to the extent that his local RJam has voted to reject GPIO Zero purely on the basis of its pin numbering (despite acknowledging that it's very nice in most other respects).

Obviously I think this is a bit crazy: why reject plenty of other advantages over the single issue of pin numbering, which arguably isn't *that* much more complex than board numbering once one takes into account the necessity of still teaching that not all pins are equal.

Nonetheless we get to discussing it, and I make the case that adding mutable global state (a la "GPIO.setmode") is a bad idea because it leaves your script open to the whims of anything you import that might change it (if anyone thinks this is unlikely: go look at early versions of picamera where I had no choice but to force BCM mode to control the camera LED because I couldn't query the mode back then; yes, it most certainly can happen).

However, I then mention the possibility that we *could* support alternate numberings anyway via an expanded definition of a pin specification. In other words, permit more than just a straight integer GPIO number to be specified. I think I'm right in characterizing his response as "excited" at this possibility :)

So, why acquiesce when I still think their position is, as stated above, "crazy"?

1. I want to encourage as many as possible to use GPIO Zero, because I'm firmly convinced it's better for teaching physical computing.
2. I don't have a magic switch I can flip to convince @gbaman to change his mind, and I judge his stubbornness to be akin to my own.
3. I *do* have the ability to implement an alternate scheme that he (and hopfully his local RJam) will find acceptable.
4. If I want to advance point 1 above, the only thing *I* (or I think *we*) can do is try point 3. So that's what I'm attempting.

I'm not proposing we remove the current integer GPIO addressing, or that we use these alternate schemes in *any* of our recipes, or encourage its use in any way (simply to minimize confusion on the part of users). As far as the Recipes chapter is concerned: we use integer GPIO addressing and that's it. However, in the PR for #459 I have added a brief bit under Notes detailing the alternate schemes available anywhere a pin specification is called for.

So, what alternate schemes have I (currently) included in the PR?

* Obviously the same old GPIO integer addressing: `LED(2)`
* GPIO and BCM prefixes for the same: `LED('GPIO2')` or `LED('BCM2')` are equivalent
* wiringPi numbering: `LED('WPI8')` or `LED('WIPI8')` are also equivalent (wasn't sure what wiringPi users usually use as an abbreviation)
* physical numbering based on the "primary" header (P1 on old Pi's, J8 on new): `LED('PHYS3')` is the third physical pin on the main header
* physical numbering based on a specific header: `LED('P1:3')`, `LED('P5:3')`, `LED('J8:3')` etc.

These are by no means "final" in my head: I think there's too many and some may not "feel" right to people. Have a play, and let me know what you think. Also, these are only used on construction. The specification is always translated internally into a GPIO number (because this is what's required at the register level, and for constructing the pin address) so you'll never see this stuff if you `repr()` a pin - it'll just read "GPIO2".

BTW, if you think this belongs in an entirely separate PR to deal with once #459 is done, I agree entirely - I'm just too lazy (and knackered!) to pull it out right now. I'll do this next week so this won't get in the way of #459.

gbaman, 2016-09-25:
Sounding great, and yes I am rather stubborn (the @NIRaspberryJam team will I am sure eagerly confirm that).
One thing to add, not sure on PHYS, is rather long and the S makes reading it with numbers (for example a 2 or 3) a little difficult.
I had rather liked the P3 syntax we had discussed, but do see that then conflicts with using that for headers. 
Guess one option could be PIN3, similar to how ScratchGPIO handled it (PIN3 for physical and GPIO3 for BCM)?
Am not right now sure of the best answer for it though...

waveform80, 2016-09-25:
> I had rather liked the P3 syntax we had discussed, but do see that then conflicts with using that for headers. 

Yeah, when I got into it I quickly ran across "hang on, P1 is a header on the earlier Pi's" so I took a leaf from wiringPi's various initializers (gpio, phys, and sys) to call it "PHYS". I agree it is a tad clunky though. I was thinking maybe "BOARD3" (to line up with RPi.GPIO's mode) ... but that's even longer.

> Guess one option could be PIN3, similar to how ScratchGPIO handled it (PIN3 for physical and GPIO3 for BCM)?

PIN feels a bit ... generic. A bit too confusing given how many other places with use the word "pin".

I kinda like PHY though. Hmmm.

lurch, 2016-09-25:
> I was thinking maybe "BOARD3" (to line up with RPi.GPIO's mode) ... but that's even longer.

That was the first thing that came to my mind when reading the first comment in this thread :)  Yeah it's a bit long, but it sounds like it's easy to support multiple 'aliases' for the same thing? (`PHY3` / `BOARD3` / `BRD3`)

And I guess the idea behind this could also be used to 'fix' #349 

Just for reference, #246 is where we'd previously discussed alternative pin-numbering schemes...

waveform80, 2016-09-26:
> That was the first thing that came to my mind when reading the first comment in this thread :) Yeah it's a bit long, but it sounds like it's easy to support multiple 'aliases' for the same thing? (PHY3 / BOARD3 / BRD3)

Hmm, multiple aliases. My internal debate has gone something like this:

"More styles?! Think of the chaos you're encouraging, man!"
"Yes, but if the whole point is to encourage people to use the library, and a style they like isn't present then haven't we failed in the endeavour?"
"But ... CHAOS!"
"Look, just because they have multiple styles at their disposal they'd still be crazy to actual use them in a single script. We can't provide foot shields for *every* possible foot-shooting scenario"
"Ah ha! But you're always saying 'code is read more than it's written'. By introducing more styles, people reading other people's GPIO Zero scripts potentially have to know more styles to understand the code, or more precisely how to wire the components associated with it"
"Okay, that's actually a good point"
"And besides ... CHAOS!"
"Shut up"

So, I think my position at the moment could be summarised: in favour of permitting (but not encouraging) multiple pin specification schemes, but also in favour of minimizing the number of aliases.

BOARD has familiarity, but not brevity ... but I think familiarity probably trumps brevity. PHY probably has familiarity for some users (wiringPi particularly uses "physical" to describe board numbering). I'm half tempted to remove the "BCMn" prefix ("GPIOn" is already permitted, and that mirrors the text of our pin-out diagram on recipes), but if we permit BOARD should we not also permit BCM as a prefix.

Urgh. The minutiae continue to fuel the internal debate.

> And I guess the idea behind this could also be used to 'fix' #349

That thought did indeed occur at the back of my mind as I was writing the translations! Basically the fixed numbering in certain subclasses within boards could simply be done with physical numbers and the board would "just work" on all Pi versions.

> Just for reference, #246 is where we'd previously discussed alternative pin-numbering schemes...

Ah, thanks for that link! It all comes flooding back ... black paint, self.kick :)

lurch, 2016-09-26:
Haha yeah, I guess there's going to be no "perfect solution" to this. CHAOS.

lurch, 2016-09-26:
...and I guess allowing pins to be addressed by their physical number, means we'll need some kind of PinDoesntSupportGPIO exception? ;)

waveform80, 2016-09-26:
> ...and I guess allowing pins to be addressed by their physical number, means we'll need some kind of PinDoesntSupportGPIO exception? ;)

Yeah, that's basically done - can't remember exactly what I added for it though (amazing how little remains in my memory from a few days ago!). Let me dig it out...

Ah, here we go - I just re-used `PinInvalidPin` for it; that's the exception that's already raised when you try and specify an invalid GPIO port (like 60). The relevant bit is here: https://github.com/RPi-Distro/python-gpiozero/pull/468/files#diff-d51cee55d16c76a12beddd2e360b8367R1106 (might need a slightly clearer exception wording there?)

That covers all the physical mappings. For example, wiringPi is basically a physical mapping so the way it's implemented is I convert from, say, WPI8 to J8:3 then recurse to convert J8:3 to a GPIO number. Likewise if you try PHYS1 it'll get converted to J8:1 then recurse and you'll hit that exception when it looks up `headers['J8'].pins[1].function` and finds it says "3V3" of "GPIOsomething"

lurch, 2016-09-26:
> Ah, here we go - I just re-used PinInvalidPin for it; that's the exception that's already raised when you try and specify an invalid GPIO port (like 60)

I wonder if it'd actually be worth having two separate exceptions? e.g. 'PHY34' could raise `PinInvalidPin` on a Model B (which only has 26 pins) but `PinNotGPIO` (or whatever) on a Model B+ (because it's actually GND).

waveform80, 2016-09-26:
Yeah, that's a decent idea - descend `PinNotGPIO` from `PinInvalidPin` and that'd be nice and clear. I might also refine the error to say something like "{pin} is a {function}, not a GPIO pin" so you'd get something like "PHYS1 is a 3V3, not a GPIO pin". Although there's a little more refinement to do there; because of the recursive lookup at the moment if you try "PHYS1" you'll actually get the error "J8:1 is not a GPIO pin".

Anyway, I'm just in the process of splitting this into separate PRs so we can refine this part a little further.

DarrenTownsend, 2016-09-26:
A couple of comments, if I may...

Do you have any thoughts as to whether @gbaman's stance (which I also think is crazy, maybe even short-sighted) is shared by many, or any, other users? I'd hate to think you're going to all this effort to appease just one person (nothing personal, @gbaman ).

I feel that you need to bear in mind where your users are coming from. I see a lot of them (as is the case with me) using GPIOZero as a first step into Python on the Pi, having used ScratchGPIO previously. I think that if you're going to go down this route, then PIN and GPIO make perfect sense, both to the end user and logically-speaking. You say, 'PIN feels a bit ... generic. A bit too confusing given how many other places with use the word "pin".' What 'other places' is it used within this context?

Anyway, hope that makes sense!

waveform80, 2016-09-26:
> Do you have any thoughts as to whether @gbaman's stance (which I also think is crazy, maybe even short-sighted) is shared by many, or any, other users? I'd hate to think you're going to all this effort to appease just one person (nothing personal, @gbaman ).

Excellent question. Okay, full justification time...

## On consistency

If I was certain it was *just* @gbaman, I'd ignore the issue and carry on (sorry Andrew!). If I was absolutely certain it was *just* @gbaman and his local RJam, and absolutely no-one else ... I'd probably still ignore it. Consistency is a useful trait in education, after all, and progress demands consensus, at least at a certain level (you don't get certain technologies without everyone agreeing the Earth is at least vaguely spherical).

Unfortunately, I'm in the awkward position of firstly knowing that there's a fairly sizeable minority of the community that *really* hate BCM numbering and are so vehemently opposed that they won't consider anything else (i.e. it's *not* just @gbaman and his local RJam), and secondly that I've played at least some part in that (although I don't think I'm directly to blame for their opposition)...

## Some history

A while back the Pi Foundation (full disclosure: I don't work for them, but I have worked with them on several occasions), decided that having two sets of standards for numbering in their educational resources was a Bad Thing, and that standardizing on one would be a Good Thing. One set of diagrams to distribute, completely consistent numbering between projects, no confusion. It would all be good!

There were strong cases made by both sides for BOARD vs BCM numbering (RPi.GPIO was the only Python implementation on the foundation's radar at the time, so the decisions were based on that, and those are the terms I'll use from here on). Incidentally, this discussion was archived *somewhere* on a GitHub ticket ... but I can't seem to find it at the moment (if anyone finds it - do please post a link here - it'd be good to refresh my memory!).

## What's in a number?

If I had to summarize, the arguments on the BOARD side boiled down to simplicity. I think it's fairly uncontroversial to say that BOARD *is* the simpler numbering. So if simplicity is your one criteria (and let's be honest, it's a damned important one for education), BOARD wins.

As you've probably guessed, I came down on the BCM side. Given this side can't win on simplicity, it needs to have some damned good counter-arguments. My recollection is too vague to be precise, so I'll just list what my current arguments for this side are (if someone can dig out that ticket, hopefully it'll show some vague consistency in my position :)

1. Not all GPIOs are physically exposed. If you want to control non-physical GPIOs, therefore, you simply can't use BOARD numbering. So the next question is:
2. Are there any use-cases for non-physical GPIOs? Two spring to mind: controlling the Pi's camera module LED (I'd released picamera by this point, so obviously this was a keen interest of mine), and controlling the Pi's internal LEDs (an esoteric use-case, admittedly, but quite useful when you want to test with some "known good" LEDs).
3. Further, BOARD numbering gives the (false) impression that any pin can be used as a GPIO pin. You still have to learn that 1 is "always on", 2 and 4 are "positively dangerous" (sorry ... bad pun ;), 6 is ground, etc. By forcing the student to learn (or at least use) the BCM numbering you make them aware that not all pins are equal.
4. BOARD vs BCM at the time (at least in RPi.GPIO) involved mutable global state. You could have one or the other. Worse still, the library allowed you to *select* the mode, but not to *query* the mode (this was subsequently fixed). Anyway, this caused particular problems in picamera because I needed to control a non-physical GPIO (for the camera LED) but had no way of knowing what mode was selected. My only option was to force BCM mode and hope I didn't blow anyone's circuit up (not a position I was at all comfortable with).

I think 1 and 2 are okay arguments, but that if you have an absolute focus on entry-level (and I'd be a hypocrite if I claimed otherwise, given our recent PyConUK presentation), they're not really applicable to total beginners. Point 3 is not very good: given that not all the pins are GPIO pins, you have to teach this in *either* system so it doesn't really argue for or against either; just that perhaps BOARD isn't as simple as you'd like.

Point 4 *was* the clincher for me. The mutable global state meant any script could arbitrarily change your pin numberings, and the inability to query it meant libraries like picamera were left with no option but to arbitrarily change it. However, that's not really applicable to GPIO Zero; we're not talking about having mutable global state. We're talking about having different specifications which are translated into BCM numbers prior to use, so no danger of imports flipping the mode, and no danger of blowing people's circuits up.

So, that's how *I* saw things. I'd guess it's how some others did as well, given how things played out. Speaking of which...

## It's a numbers game

I think I'm right in saying that the BCM side was more popular on the numbers ... but there were a significant number on the BOARD side. Over the next year (ish?) I was involved as a tutor at several Picademy sessions, and occasionally had discussions with attending teachers regarding BOARD v [... cut ...] xisting declarations (unlike global state), and this final argument imagines a user *already versed* in GPIO Zero, whilst the intended beneficiary here is the complete beginner.

For the existing GPIO Zero user, this change makes no difference to their existing scripts and simply means that in future they may encounter a few odd scripts using different pin specifications. For the beginner, it may mean the difference between them encountering GPIO Zero in their education, or not.

I rest my case.

lurch, 2016-09-26:
> You say, 'PIN feels a bit ... generic. A bit too confusing given how many other places with use the word "pin".' What 'other places' is it used within this context?

Pretty much *everywhere* in the documentation (and the code), e.g. https://gpiozero.readthedocs.io/en/v1.3.1/api_output.html#led
And in https://gpiozero.readthedocs.io/en/v1.3.1/api_pins.html

Like @waveform80 I'd definitely be :-1: on having "PIN" as an alias for the physical-pin-numbers. (But then, I've never used ScratchGPIO)

lurch, 2016-09-26:
Woah, nice essay @waveform80 :-D
(it was posted at the same time as my previous comment)

Another little point about RPi.GPIO - when the Rev2.0 Model B Pis came out with both a P1 and P5 header, you **had** to switch to `BCM` numbering mode if you wanted to control any of the GPIO pins on the P5 header, `BOARD` mode can only control pins on the P1 header.

Out of curiosity, what does @gbaman and his @NIRaspberryJam think of http://rasp.io/prohat/ (which uses 'BCM' numbers) ?

waveform80, 2016-09-27:
Okay, I've now got the PRs separated. So, there's the one for remote SPI (#468) which we really ought to review and deal with first. I don't think it's contentious, but it is horrifically long, and the pin numbering PR (#470) branches off it so remote SPI really has to get merged first.

> Like @waveform80 I'd definitely be :-1: on having "PIN" as an alias for the physical-pin-numbers. (But then, I've never used ScratchGPIO)

Yeah, I'm afraid PIN is waaay too vague given all the uses of it throughout the docs and code. Sorry!

Having had a think about this, I reckon a decent approach would be: don't dive in with a ton of aliases. Let's start with one prefix for each "numbering system" and see if that brings the hold-outs on board. Adding more aliases later is easy enough, and doesn't break compatibility whilst adding too many up front and then removing them later would break compatibility.

So ... which prefixes? Here's a set I could live with:

* 'GPIOn' for straight GPIO numbering (i.e. just the current system with a prefix). It's perfectly descriptive of what the pin represents, it mirrors the pin-out diagram's labels, etc.
* 'PHYn' for physical (board) numbering. It mirrors (ish) what wiringPi calls physical numbering.
* 'WPIn' for wiringPi numbering (seems like WPI is a more typical contraction of wiringPi than WIPI after I trawled the forums a bit).
* 'h:n' for explicit header:number; this really has to be present in any system as it's the basis for all physical numberings

Here's another set I could live with, but I *think* I prefer the first set at the moment:

* 'BCMn' for straight GPIO numbering. This obviously mirrors RPi.GPIO so this set is fairly obviously appealing to existing RPi.GPIO users.
* 'BOARDn' for physical numbering. Because this set includes 'BCMn' it makes sense to offer the similar prefix for the physical scheme
* 'WPIn' for wiringPi numbering. No change here.
* 'h:n' for the same reasons as above.

Anyway, for now let's concentrate on reviewing and merging the remote SPI stuff. Once that's done, the PR for this will be nice and short and I can tweak it to fit a minimal system like one of the above (or whatever we can all vaguely agree on). We won't please everyone with this, but we don't have to please them ... we merely need to appease them sufficiently ;)

lurch, 2016-09-27:
> seems like WPI is a more typical contraction of wiringPi than WIPI

Yeah, IMHO WIPI sounds too much like WiFi, and there's also http://uk.farnell.com/element14/wipi/dongle-wifi-usb-for-raspberry/dp/2133900

My only concern with the current prefixes is that it might be easy to mis-read `WPI2` as `WP12` ? (and could `GPIO5` be mis-read as `GPI05` or even `GP105` ?) Think about people in workshops (e.g. @NIRaspberryJam ) typing in code from printouts...
One way to mitigate that _might_ be to put a separator between the prefixes and the numbers, so that it's clearly only-letters-before-the-separator and only-numbers-after-the-separator e.g.
  * 'GPIO:n'
  * 'PHY:n'
  * 'WPI:n'
  * 'PHY:h=n' (or 'PHY(h):n')

but that obviously risks over-complicating it again :-/
Would a stringified-number i.e. `'n'` be interpreted as just an integer `n`?

The other minor concern I have is that people might expect code using 'P1:11' to work on all models of Pi, whereas I guess it'd have to be changed to 'J8:11' to work on the B+ onwards, with the way you've currently got things?

joan2937, 2016-09-27:
I'd just provide some translation functions.

So if method x took two GPIO.

x(7, 11) # Broadcom GPIO 7 and 11
x(BOARD(26), BOARD(23)) #  Broadcom GPIO 7 and 11
x(wPi(11), wPi(14)) #  Broadcom GPIO 7 and 11

DarrenTownsend, 2016-09-27:
Wow, that's definitely an in-depth explanation Dave, thanks.

I think I would probably agree that BCM numbering should be the preferred method (definitely from KS2 onwards). As you say, most resources now use that system, and ScratchGPIO can use either. I've just started a new group of kids and I was planning to use pin numbers, but I think I'll try them with BCM tomorrow and see if they grasp it ok.

I like Joan's suggested format, seems quite tidy. If those are the 3 methods, would it be viable to just use G,P(or B),W as prefixes or would that require another level of explanation beforehand?

lurch, 2016-09-27:
I suspect that @joan2937 's solution (while it looks neat) wouldn't work with remote-pis being controlled by pigpio?

lurch, 2016-09-27:
...although I guess if somebody is vehemently against the 'PHYn' numbering, and is insistent on being able to use the 'PINn' notation from ScratchGPIO, I guess they _could_ do:
```python
def PIN(n):
    return 'PHY%d' % n
```
at the top of their script, and just use e.g. `PIN(5)` everywhere in GpioZero that accepts 'PHY5' ;-)
(although we probably shouldn't encourage this)

cymplecy, 2016-09-28:
My tuppence
I must first declare that I've even more for pin numbering than @gbaman :)

But ignoring the big mistake made in a room in Cambridge a long time ago..... 

it would be great to let users choose what numbering system to employ in GPIO.zero

However, introducing another word for pin when they are universally known as pins is just not worth it

it'd only be worth it if ubergeeks wanted the pin numbering scheme and they generally don't.

Its the KISS camp that wants/needs pin numbering.

Pin:  Do or do not- no compromise there is

Even though I want pin numbering with every fibre in my body - I wouldn't want ANYONE to have to go thru the whole Calcutta/Mumbai  Peking/Beijing changes word use change.

but why am I bothering saying anything ...... what do I know........ :-)

lurch, 2016-09-28:
Innocent question (from a self-confessed ubergeek) : When things like http://rasp.io/prohat/ and http://rasp.io/portsplus/ and http://pinout.xyz/ and https://www.raspberrypi.org/blog/raspberry-leaf/ ( https://github.com/splitbrain/rpibplusleaf ) are available, why are people still **so** resistant to the BCM numbering scheme?

joan2937, 2016-09-28:
@cymplecy @lurch 

It possibly comes down to the difference between being a developer and being a consumer.

From a developer point of view I didn't have to give any thought to knowing that Broadcom numbering was the appropriate way of defining the GPIO.  All the developer documentation is based on Broadcom GPIO numbering.

From a consumer point of view I'd prefer pin numbering - especially now that the 40 pin expansion header seems stable.

For people who have a foot in both camps like cymplecy and gpiozero it's a bit of a conflict.

waveform80, 2016-09-28:
@lurch 
> Yeah, IMHO WIPI sounds too much like WiFi, and there's also http://uk.farnell.com/element14/wipi/dongle-wifi-usb-for-raspberry/dp/2133900

Okay, WIPI's definitely out. WPI's on the list for now. I'll address "W" as a sole prefix a bit below.

> One way to mitigate that might be to put a separator between the prefixes and the numbers...

I'm not *terribly* worried about 0 vs O issue. They must already know the library is "GPIO Zero" or more precisely gpiozero otherwise they wouldn't be able to import it in the first place. We don't use leading zeros in the pin-out diagrams or any prose, so I don't think there's great danger there. As to people using them (`"GPIO04"`): the code is fairly tolerant - it will happily ignore leading zeros if you happen to use them (but we shouldn't encourage their use - having O and 0 next to each other is pretty awful practice).

> Would a stringified-number i.e. 'n' be interpreted as just an integer n?

The current code doesn't (it'll just throw an error), but I wouldn't be averse to adding that case if it makes sense?

@joan2937 
> I'd just provide some translation functions.

I wish it were that simple. Admittedly at the moment, it is. However, consider the future of supporting IO expanders which incorporate their own (potentially non-sequential) GPIOs (and other sorts of pins). How is the function to know it is being called for the main header, or for an expander, and thus what translation to perform?

@DarrenTownsend 
> I like Joan's suggested format, seems quite tidy. If those are the 3 methods, would it be viable to just use G,P(or B),W as prefixes or would that require another level of explanation beforehand?

When chatting with @gbaman at PyConUK we were initially discussing Pn, Bn, etc. as a simple scheme. Unfortunately the early Pi models labelled their headers P1 and P5 which instantly throws some confusion into the Pn scheme. The later Pi's use J8 as the primary header's label (J is a more "normal" prefix for this sort of thing).

At the end of the day, if it comes down to a decision between "unambiguous, but longer to type" and "slightly ambiguous, but shorter to type" I'm going to side with the unambiguous option because I'm of the opinion that while having less to type marginally improves ease of use, it's more important that the code is easy to *understand* than that the code is slightly easier to *type*.

Sorry - will get back to the rest of the comments later!

cymplecy, 2016-09-28:
"Innocent question (from a self-confessed ubergeek) :why are people still so resistant to the BCM numbering scheme?"

1.  Kids aren't uber-geeks :)
2.  Pin counting doesn't require any learning at all - just counting.  There is a 1:1 correlation between software and hardware that doesn't require ANY explanation (apart from zig-zagging down the connector of course :)

The problem is that uber-geeks actually prefer BCM and are 99% genetically incapable of seeing that it wasn't the way to go forward with teaching resources and lesson plans :)  They weren't able to perform pin abstraction :)

But that war was lost long ago and I assume shortly that @waveform80 will find a "friendly" hand leaning on his shoulder saying - forget about pins :)

But - lets see how far this goes :)

lurch, 2016-09-28:
@waveform80 

> > Would a stringified-number i.e. 'n' be interpreted as just an integer n?

> The current code doesn't (it'll just throw an error), but I wouldn't be averse to adding that case if it makes sense?

Well, we already do so for the pi_info() function, and IMHO it wouldn't hurt?
Although off the top of my head I can't see any reason why you'd do `LED('5')` instead of `LED(5)`. Unless of course you're getting the GPIO numbers from some external source, and `LED(num)` might look neater than `LED(int(num))`.

> I'm going to side with the unambiguous option

Agreed.

@cymplecy 

Do kids ever have problems with the "zig-zag counting"? I guess taking the pin-counting-approach to the extreme (Devil's advocate) would end up with two separate single-row headers (a la Arduino). Which in a way is the route that http://rasp.io/prohat/ took.
Ping @raspitv as he had some good comments the last time we were discussing pin-numbering.

And I guess if they're using a case like https://shop.pimoroni.com/products/pibow-coupe-for-raspberry-pi-3 you have to explain that they need to _ignore_ the numbers on the case ;-)
Given that BCM and BOARD pins numbers are both widely used, I guess even if you *only* teach BOARD numbering, you might have to explain that BCM numbering also exists?

> They weren't able to perform pin abstraction :)

@waveform80 _does_ explain above why it wasn't possible to 'universally' use the physical-pin-number abstraction everywhere (which Arduino is able to, where "Digital Pin 10" is _really_ "PB2" under the covers).

> But - lets see how far this goes :)

Yup :)

gbaman, 2016-09-28:
@lurch in relation to why are people still resistant to BCM with all the other boards etc, the point is a kid at our jams shouldn't need one of those boards? I don't want to spend ~£500 on a set of portsplus's, why should we have to when 99.9% of the work they will do will simply use the GPIO pins available as opposed to other weird pins, like camera LED or the extra header on the old Pis.
The kids at our jams won't touch either, basically ever. The advanced uber geeks may someday, but putting them off at the start with even more confusing points right when they start out is *not* what me or my Raspberry Jam team want to do.    

Sure if you have geeks, then fine, explain to them what BCM is and why, or get their parents to buy a £15 board. **But for those beginners** (which we get a heck of a lot of), abstracting it out like every other board on the planet does (for good reason) is a far better approach. **Is this not what gpiozero came along to do in itself?** Abstract out a complicated thing for beginners? Why should the 9 year old I am teaching to blink an LED care about what is below? They should just be able to count, instead of looking up a diagram or using an addon board to figure out the number of the pin they want to use from a seemingly random order.

@waveform80 the initial discussion from 2 years ago https://github.com/raspberrypilearning/push-button-stop-motion/issues/5

raspitv, 2016-09-28:
@gbaman £500 will get you 500 portsplus at wholesale rates. I don't think you need quite that many ;p Incidentally Portplus in its current incarnation has BCM on one side and Pin numbers on the other so you don't need to count when you're using BOARD either. But apparently some people mess up and use the wrong one sometimes. GAH!

I've always used BCM because that was what was on the first board I learnt with - the Gertboard. It's going to be the same for anyone learning. Whatever you learn is "correct" for you and you'll probably want to stick to it. Some favour one, some favour the other and it's actually pointless arguing about it because there will never be agreement.

But there is a strong case, if one scheme is strongly favoured, for only supporting that scheme.
(Just thinking out loud, not saying that's my position on the matter.)

My bugbear is the use of the terminology "pin 14" instead of "GPIO 14" in documentation, because to a BOARD user pin 14 is physical pin 14, but to a BCM user, pin 14 is GPIO 14. That is why I advocate the use of the word "port" in all my RPi.GPIO documentation when I'm talking about ports, and "pin" when I'm talking about physical pins. Makes sense to me. YMMV, but people seem to find my docs fairly clear.

I also hate counting pins and think it is prone to wiring errors. That's why portsplus exists.

gbaman, 2016-09-28:
Whoops @raspitv, had meant http://rasp.io/prohat/ which @lurch had mentioned few posts back :)

lurch, 2016-09-29:
>  I don't want to spend ~£500 on a set of portsplus's

That's why I also mentioned the 'leaf' printouts ;-)

> Whatever you learn is "correct" for you and you'll probably want to stick to it.

True. See also Mac vs. PC... ;-)

> That is why I advocate the use of the word "port" in all my RPi.GPIO documentation when I'm talking about ports, and "pin" when I'm talking about physical pins. Makes sense to me.

Yeah, given that everyone is coming from a different direction, and has their own preferences, I guess there's no perfect terminology. "port" could also relate to USB port, Ethernet port, HDMI port, etc.
(I wonder if that's where the 'P' in "header P1" came from?)

> I also hate counting pins and think it is prone to wiring errors.

...and surely if you're counting pins, you're more likely to count "5 from the left on the top row", or "3 from the right on the bottom row"? Which of course doesn't have any direct relation to BCM **or** BOARD numbering.

I guess part of the approach that myself and (perhaps) @waveform80 are looking at is "there are some things you can *only* do with BCM numbering, and (hopefully) some users of GpioZero may progress from basic to advanced usage, so it makes sense to just teach them one numbering scheme (BCM) from the beginning, rather than having a logical disconnect when the user progresses from basic to advanced usage" ?
Whereas the approach @gbaman is looking at is "the people coming to my Jam will always be basic users, and will **never** need to know about advanced usage" ?
(but again perhaps that's just my ubergeek PoV)

Just to get some more input, I wonder if @Gadgetoid (author of http://pinout.xyz and https://github.com/WiringPi/WiringPi-Python) has any thoughts? :)

Apologies if I'm coming across as antagonistic - that's not my intention, I'm just trying to understand people's reasoning.
I fully support the multiple-ways-to-specify-pins that @waveform80 is suggesting in this issue. (apologies if I've accidentally dragged this discussion too far off-topic from the original intention!)

WayneKeenan, 2016-09-29:
Hi All!

It would like to request that GPIOZero supported (at least a basic subset of) board agnostic pin names/aliases, and by ‘board’ I’m not limiting gpiozero to Pi boards.

"Why?"  Well..

Because, I’ve ported (a subset of) gpiozero and also the Python BluetoothLE module ‘bluezero’ to a port of MicroPython for: nrf51 (e.g. micro:bit),  nrf52 (new Arduino’s, RuuviTag and more) devices.

With the port I’ve created it’s possible to write Python code that can control an LED via BluetoothLE and that the Python code will run unmodified across:  Raspberry Pi’s, micro:bit’s, Arduino’s (ARM-cortex devices) and many Nordic BLE SoC based devices.
I’ve had interest from various educational related folks and it would be very convenient from a learning. documentation and maintenance perspective, to be able to create many more types of example that would not need modifying, because of variations in board pinouts, by using these aliases.
I’m not expecting to be able to cover every possible gpio use cases, but the basics to support simple experimentation on different sorts of hardware should be easy.

There are some examples for the current implementation here:  https://thebubbleworks.com/bubblepy-examples

For what I’m requesting the GPIOZero implementation (for the board gpiozero is installed on) would map these (example only) names to board specific pins at runtime:

D0 .. Dn			(digital)
A0 .. An			(analog)
LED_0 .. LED_n
LED_BLINK_PIN	(e.g.  maps to 13 on Arduino ARM based boards, pin 17 on some BLE modules, etc.)
BTN_0 .. BTN_n
SDA		(default)
SCL		(default)
SDA_0 .. SDA_n
SCL_0 .. SDC_n

ADC0 would have no meaning on a Pi    (unless a default add-on is assumed or has been setup perhaps, a seperate 'device registry’ topic perhaps.)

Also, could pi_info() be aliased as board_info() please? That would also be vurra nice too :)

Thanks
Wayne

DarrenTownsend, 2016-09-29:
lurch said "...although I guess if somebody is vehemently against the 'PHYn' numbering, and is insistent on being able to use the 'PINn' notation from ScratchGPIO, I guess they could do:

def PIN(n):
    return 'PHY%d' % n
at the top of their script, and just use e.g. PIN(5) everywhere in GpioZero that accepts 'PHY5' ;-)
(although we probably shouldn't encourage this)"

Is this any different from 'import StupidlyLongName as Name' ? Why would it be frowned upon?

@gbaman "Sure if you have geeks, then fine, explain to them what BCM is and why, or get their parents to buy a £15 board. But for those beginners (which we get a heck of a lot of), abstracting it out like every other board on the planet does (for good reason) is a far better approach. Is this not what gpiozero came along to do in itself? Abstract out a complicated thing for beginners? Why should the 9 year old I am teaching to blink an LED care about what is below? They should just be able to count, instead of looking up a diagram or using an addon board to figure out the number of the pin they want to use from a seemingly random order."

THIS!

I think it all boils down to who you feel your users are. In my opinion, it is ideal for Upper KS2/KS3 moving on from Scratch on the Pi and making their first steps in Python, or using it because they want to do something that is not possible, or too cumbersome, in Scratch.

I also think that there is a danger of going the same way as NuScratch, in trying to build in support for every add-on out there, but forgetting that your target market (kids, schools) does not have the money or need for these extras.

lurch, 2016-09-29:
>> (although we probably shouldn't encourage this)

> Is this any different from 'import StupidlyLongName as Name' ? Why would it be frowned upon?

I was only suggesting that it shouldn't be encouraged, because the more different naming schemes that are being used, the harder it is to copy'n'paste snippets of code out of other peoples' gpiozero scripts into the gpiozero script you're currently working on. (which seems to be the way lots of beginners operate)

> I also think that there is a danger of going the same way as NuScratch, in trying to build in support for every add-on out there, but forgetting that your target market (kids, schools) does not have the money or need for these extras.

Not sure I follow? Just because something *supports* a particular feature or add-on board, doesn't mean you have to *use* that feature or add-on board?

DarrenTownsend, 2016-09-29:
> Not sure I follow? Just because something supports a particular feature or add-on board, doesn't mean you have to use that feature or add-on board?

Agreed, as long as trying to provide support for some random add-on, or trying to develop something that suits everybody (even with the best intentions), doesn't end up alienating your primary audience.

recantha, 2016-10-04:
For what my opinion is worth (which in coding GPIO Zero is nothing, but in terms of using the thing _might_ be worth something)... I know it might be a royal PITA to implement, but having both numbering systems just seems like the right thing to do. It certainly won't affect take-up negatively, and think of all the possible positive effects it might have. Just imagine, having BOARD numbering in there might _just_ encourage someone who is on the borderline between enthusiasm and abject terror to try stuff out. I'm not advocating that anyone should use anything other than BCM, God forbid, but if counting pins is what we need to engage those with less ability to cope with the abstraction, go for it! 👍 
Love what you're doing with GPIO Zero, guys. 😀

lurch, 2016-10-04:
"someone who is on the BOARDerline..." ? ;-)
(sorry)

waveform80, 2016-10-22:
@recantha - yup, that's pretty much my stance (as hopefully comes across from my excessive verbiage above).

@WayneKeenan - right now I'd like to concentrate on just getting the alternate specs in for the Pi. As mentioned above I'd rather do minimal additions now and leave room to add more later than add too much now and have to look at removing stuff later. On `pi_info()` vs `board_info()`: it is something that crossed my mind when constructing the pin factories, but I'm not sure what form the result should take if it's to cover a wide variety of boards in the future. So for the time being, I'm going to stick with `pi_info()` which returns a very pi-specific structure; in future we may we add a more generic `board_info()` method which returns something more generic (and re-factor `pi_info` so it becomes a stub that calls that, or something else). It is something we're thinking about though!

To everyone else - I haven't forgotten about this, but I need to get the remote SPI stuff in first as this PR builds on that one. @lurch has very kindly spending some time reviewing that PR (and picked up several things I've been addressing); anyway, once that's complete I'll rebase this one and we can get something merged.

Gadgetoid, 2016-10-22:
I've done some reading up, but I can't help but feel like I'm a little behind the curve still so forgive me for trundling in like a blind elephant and trumpeting out my ill-informed opinions.

Both WiringPi and RPi.GPIO demand that you "set up" the GPIO before use, explicitly declaring which numbering scheme you're going to be using. What's the reason for using constant-defined, or string-named prefix schemes versus just doing this? Even in documentation, it's easier to communicate "This uses scheme X" and then simply write the numbers as numbers, than mangle everything with prefixes. A prefix system only serves to encourage people to *mix* different schemes, which would be a sin much worse than using WiringPi numbering :D 

TLDR: What's the technical reason for:

```
do_stuff_with_pin(GPIO14)
```

versus:

```
use_scheme(GPIO)
do_stuff_with_pin(14)
```

As for the merit of physical numbering; as much as "counting pins" may be the path of least resistance, I'd venture that it's not a method which has any real practical use beyond the Pi. I've found few micro-controllers, for example, where the designer has felt it prudent to number pins in any fashion other than apparently at random.

Being able to read, understand and relate to a pinout diagram and translate that information to a physical piece of hardware is, in and of itself, a valuable skill. Supplying text instructions to communicate the location of pin 1, and the practise of counting pins is, arguably, as convoluted, but less widely applicable than such a skill. Of course, as one of the pinout.xyz maintainers I'm a little biased :P 

I'm particularly interested in the nomenclature being put forward here, since pinout.xyz should reflect the consensus of the community.

waveform80, 2016-10-22:
> TLDR: What's the technical reason for:...

Mutable global state. I've covered it above, but I admit this thread's too long to skim. The issue with making pin numbering dependent upon global state is that innocently importing any module which tampers with that state can break your script (not to mention your circuit).

In case anyone thinks "surely that's unlikely to happen" (i.e. I'm just spouting hypotheticals) bear in mind that historically one couldn't query RPi.GPIO's numbering mode so in older versions of picamera where I needed BCM mode to control the camera LED (because it's a non-physical GPIO so you can't address it from BOARD mode) I had no option but to force the mode to BCM and hope I didn't blow up anyone's stuff. In other words, this isn't just theoretical. People can, and have, simply forced the GPIO numbering scheme in libraries that users are likely to import (without necessarily reading the docs and the associated [warning boxes](http://picamera.readthedocs.io/en/release-1.12/recipes1.html#controlling-the-led)).

This also means that people can copy'n'paste between scripts which use entirely different numbering schemes and things will still work (although their code will have a distinctly schizophrenic feel to it).

Still, that's a minor benefit: the major objection is mutable global state. It's evil :)

Gadgetoid, 2016-10-23:
> Mutable global state.

I should have guessed that one. Also it was in the first post. I'm a tool ;)

I can totally understand why that's a no no. An unfortunate side-effect of being able to use the `GPIO` module as imported, rather than having boilerplate where you create a class instance.

I think strings are ugly. If numerical pin addressing is still supported, I suppose there's nothing to stop me creating "constant" definitions like: `GPIO2 = 3` and `GPIO3 = 5`. Sadly Python doesn't make importing a set of constants particularly compelling, either `from Whatever.Pins import *` or import Whatever` and then `Whaever.Pins.GPIO2`. Ugh.

Anyway, I'm just mumbling. My main reason for being here is Pinout, and how it could be changed to better reflect the "one true way."  I shall make a note of this thread!

gbaman, 2017-06-24:
Is there any further update on this? We are still keen to use GPIO Zero at the NI Jams, but as a team we made the decision that for the activities we do, Physical layout is far better for our kids so are unable to use GPIO Zero as of yet.
We are in the process of designing activities for next term and it would be perfect if we could use GPIO Zero.

DaddyTheRunner, 2017-06-25:
@gbaman and waveform80 I read through a good portion of this thread last night and I appreciate the challenges presented. 

What I don't understand is why isn't @gbaman's use case easily solved with a set of variables that maps physical pins to BCM integers? From a beginner's and teaching point of view, it's no different than defining variables for RGB values when introducing color. Think RGBLEDs.

The list of variables wouldn't even have to be part of the official library. They could be in a stand alone module that gets loaded by groups wanting to use board numbering. 

What am I missing?

filips123, 2017-11-03:
If you were to support both ways, but one would be default, it would not be more complicated.
In addition, I am interested in how to support the older versions of Raspberry Pi, which have a different labeling without changes.

closed 2019-01-24 by bennuttall

lurch, 2019-02-13:
Great news - [GPIO Zero v1.5](https://www.raspberrypi.org/blog/gpio-zero-v1-5/) now supports [BOARD](https://gpiozero.readthedocs.io/en/stable/recipes.html#pin-numbering) numbering :grinning:

commit b62ecd0e (2019-01-24): Argh its pin numering again (#470)

* Fix #467

Introduces alternate pin specifications (i.e. physical numbering
systems)

* Tidy up pin numbering branch

Get rid of some of the aliases and consolidate all pin spec parsing in
one place.

* Use BCM/BOARD/WPI

After discussion with Ben, we've decided that despite BOARD being overly
long this represents the path of least resistance for users transferring
from the most popular GPIO library (RPi.GPIO). GPIO is left as an alias
for consistency with `repr` output and WPI is included for wiringPi
users.

* Rebase on master and fixup tests

* Add GPIO1 [... cut]