gpiozero/gpiozero #471: Documenting source tools: challenge

waveform80, 2016-09-27:
Here's a challenge for someone to take on (I've had it in the back of my head for a while after adding charts to the soft SPI protocol docs): I'd love some graphs in the tools docs. It's all very well describing the behaviour of something like [`absoluted`](http://gpiozero.readthedocs.io/en/v1.3.1/api_tools.html#gpiozero.tools.absoluted) in prose but wouldn't it be so much clearer if, with the docs, there could be a nice ASCII-art chart embedded to *show* you the result? Example:

```
:func:`absoluted` transforms the following wave::

    +1 T
       |          ,--.
       |          |  |
       |          |  |
    0  +--.  ,----'  `---.     ,----|
       |  |  |           |     |
       |  |  |           |     |
       |  `--'           `-----'
    -1 _

Into this::

    +1 T
       |  ,--.    ,--.   ,-----.
       |  |  |    |  |   |     |
       |  |  |    |  |   |     |
    0  +--'  `----'  `---'     `----|
       |
       |
       |
    -1 _
```

Now the above is fairly simple, and to come up with something that just handles 1s, 0s, and -1s in a style like that isn't terribly hard - probably about a day's work to do it nicely. If you just want to try that, great! But here's a harder challenge for the crazy brave: make me something that'll handle an *arbitrary* waveform so I can use it properly for stuff like `booleanized` with hysteresis set.

Why ASCII art? Yeah, I could use an image, but then you can't see it using Python's help() at the command line. And besides ... I love ASCII-art :)

lurch, 2016-09-27:
> arbitrary waveform...

```
 :func:`absoluted` transforms the following wave::

    +1 T
       |          ,--.
       |          |  |
       |          |  |
    0  +--.  ,----'  `---.             ,--|
       |  |  |            \       _   /
       |  |  |             \     / \_/
       |  `--'              `---'
    -1 _

Into this::

    +1 T
       |  ,--.    ,--.      ,---.   _
       |  |  |    |  |     /     \_/ \
       |  |  |    |  |    /           \
    0  +--'  `----'  `---'             `--|
       |
       |
       |
    -1 _
```
;-)

IMHO this is pushing ASCII-art too far (I think if you were trying to show hysteresis, ASCII is too low-res), and I think using images would be better.

waveform80, 2016-09-27:
Yeah, that's the sort of thing. And yes, you're probably right (especially given display font variants) ... but I'd love to try it anyway :) I was half-tempted to have a dive into libcaca or aalib earlier, but I haven't got time at the moment!

waveform80, 2016-09-27:
Ah! gnuplot 5 supports libcaca ... well, that's this issue mostly solved when it trickles downstream ;)

waveform80, 2016-09-27:
https://gist.github.com/waveform80/a5bdbf13a2f56f9af95c2472326056fb

lurch, 2016-09-27:
> especially given display font variants

That's a good point - what happens if someone's got their terminal set up with a proportional-width font??

I can't be bothered to test your gist right now - do you fancy posting a 'screenshot'? ;)

waveform80, 2016-09-28:
> That's a good point - what happens if someone's got their terminal set up with a proportional-width font??

If someone's using proportional fonts on a terminal I'd declare them clinically insane and back away slowly! I was thinking more about variation in mono-space fonts (i.e. does tilde appear at the top of a character box, or the middle and so on).

> I can't be bothered to test your gist right now - do you fancy posting a 'screenshot'? ;)

Sure thing! I'll update the gist with it... done. Basically it just presents a little scrolling graph. If you wire up a button on GPIO17 and bash it, the graph shows the "value" of the button scrolling by. Very crude - doesn't have variable size axes, doesn't allow the axis max and min to be set, etc. etc. but it's quite nice for a few minutes work.

lurch, 2016-09-28:
> If someone's using proportional fonts on a terminal I'd declare them clinically insane and back away slowly! 

LOL! :)

> Sure thing! I'll update the gist with it... done.

Oooh, nice. I guess it's probably a little too esoteric to include in the Recipes page though? ;-)
Maybe it could detect 'rapid' button presses (using the `when_pressed` callback) that occur between updates, and display them as:
```
 |
 |
 |
-+-
```
And the maxlen of your deque should probably be reduced to 75. Not that I'm condoning this as a good idea...

waveform80, 2016-09-28:
Heh, yeah - it's a little too complex for the recipes page (what with transpose, pairwise and all that). And yes, the One True Interface does indeed have 80 columns so I ought to reduce the deque's maxlen (although I confess my terminal is rather wider given I rarely fire up X and the Pi has a 1080p monitor).

Might have to have more of a play with this later on (unfortunately I pulled my back out whilst trying to pick up our daughter last night, so I'm currently indulging in some recreational coding to take my mind off the throbbing agony at the base of my spine!)

lurch, 2016-09-28:
Ouch! Hope you're feeling better soon. Sounds like you need to feed your daughter less cake ;)

waveform80, 2016-09-28:
Hehe - she is a sucker for cake (but then I'm not sure many 1.5 year olds aren't :). Stupidly I bent instead of crouching ... eurgh. It's just muscular though, nothing serious thankfully!