gpiozero/gpiozero #472: Start using changelog

thijstriemstra, 2016-09-27:
Would be nice to have a document in the repository that describes the changes between releases.

bennuttall, 2016-09-27:
You mean like https://github.com/RPi-Distro/python-gpiozero/blob/master/docs/changelog.rst and http://gpiozero.readthedocs.io/en/v1.3.1/changelog.html ?

waveform80, 2016-09-28:
Heh - I am quite a stickler for changelogs. I hate it when changelogs are broken into multiple files (so I can't just search across the entire history), when they don't include dates (so I don't know whether the project's still alive or not), when they don't link to an issue tracker (so I can read up on all the detail behind a change I'm particularly interested in), etc.

I'm afraid my process for writing them is a bit manual though. Largely consists of going to GitHub issues just before a release, filtering on closed tickets for the current milestone and manually writing the changelog from that list. Unfortunately I've never found a *really* good changelog generator (or found the time to write one :).

bennuttall, 2016-09-28:
Maybe we could add a label to issues which will add a new feature, or make a significant API change, and use those labels, then use those to filter issues to show changelog-worthy changes?

waveform80, 2016-09-28:
Possibly - I generally try and make it as comprehensive as possible but I do filter out some labels (e.g. stuff tagged "question" usually doesn't make the changelog while anything marked bug or enhancement does)

thijstriemstra, 2016-09-28:
Ah damn, missed that doc. Maybe it should be linked from the readme.

bennuttall, 2016-09-28:
It's in the sidebar on the documentation.

closed 2016-09-28 by bennuttall

lurch, 2016-09-28:
> I hate it when changelogs are broken into multiple files (so I can't just search across the entire history), when they don't include dates (so I don't know whether the project's still alive or not), [...]

Here's an example that hits all 3 of those: https://www.pureftpd.org/project/pure-ftpd/news

waveform80, 2016-09-28:
> Here's an example that hits all 3 of those: https://www.pureftpd.org/project/pure-ftpd/news

*shudder* Yeah, that's exactly the sort of thing I'm talking about. When was that released. Last week? Last year? 5 years ago? It's under "Latest news". If it's "news", *put a damned date on it*! I can forgive not having a combined log if you've got a really huge history, but at least make it easy to find the other entries (if that's "latest news", where's "older news" or "positively ancient news"?). Postgres is a good example of a project with [*huge* changelogs](https://www.postgresql.org/docs/9.6/static/release-9-5-4.html), but note how they've got the release date front and centre; plus, pop up one level and you get the complete changelog right back to postgres95. They don't have links to their bug tracker, but at least they do have links to their docs so if any change looks interesting I can immediately go and read up on it (or bookmark it for a later perusal).

> Ah damn, missed that doc. Maybe it should be linked from the readme.

That's a decent idea - if you want to know what's new and you're not aware of the changelog what's the first place you go? GitHub. And what's the first thing it displays? The README. Hmm, might not hurt to have a changelog link in there...

lurch, 2016-09-28:
@bennuttall 
> It's in the sidebar on the documentation.

@waveform80 
> GitHub. And what's the first thing it displays? The README. Hmm, might not hurt to have a changelog link in there...

Is that yet another reason to have a separate 'index' file used for GitHub and ReadTheDocs? ;-)

waveform80, 2016-09-28:
> Is that yet another reason to have a separate 'index' file used for GitHub and ReadTheDocs? ;-)

Yeah, I'm more or less convinced at this point that GH's ReST rendering is sufficiently bad that trying to avoid having two intro files is pretty futile.

thijstriemstra, 2016-10-02:
> The README. Hmm, might not hurt to have a changelog link in there...

See PR #475.

lurch, 2016-11-28:
@waveform80 

> Unfortunately I've never found a _really_ good changelog generator (or found the time to write one :).

https://github.com/resin-io/versionist is attempting to solve that problem - maybe it's something you'd be interested in looking at?