gpiozero/gpiozero #478: color attribute for LED class

thijstriemstra, 2016-10-02:
I'm porting my app to python-gpiozero. My old `Pin` class is similar but also accepts a title and color, e.g.:

```python
RED = QColor(237, 33, 54)
GREEN = QColor(83, 237, 89)

led1_pin = Pin(15, 'System Error', RED)
led2_pin = Pin(19, 'System OK', GREEN)
```

Now I see that the `LED` or `PWMLed` doesn't have a label or color attribute. But every LED has a color right? Not being able to add a label is not as big of a deal, but having a color would be nice. Do you think this is useful to have or should I subclass LED and these attributes myself?

lurch, 2016-10-02:
From your example code, it seems you're still using a "pin-based approach". In GpioZero we try to use a "component-based approach". So your example would become:
```python
from gpiozero import LED

red_led = LED(15) # or just 'red = LED(15)'
green_led = LED(19)
```

Could you give an example of how your `title` and `color` attributes would be used?

thijstriemstra, 2016-10-02:
@lurch I have built a small tool with Qt that allows you to control/test the various LEDs (enable disable one or more at the same time) and I use the color attribute to give the led icon a different color, and the label to describe the LED.

[image]

I can imagine the label attribute not being useful for gpiozero but the color attribute can be useful, or makes sense to me, just like the color on RGBLED. An RGB led can show any color but a simple LED also has a color, although only one. I've already subclassed PWMLED to add these attributes (also had to a `setup` method that actually instantiates the PWMLED because I'd get OSErrors on Mac OSX because the `NativePin` could not be created).

thijstriemstra, 2016-10-02:
So I guess having the color value of any LED in the repr of the LED object would be nice. It doesn't have to be a valid color either, something like 'red' or any string would be ok.

```python
>>> led = LED(12)
>>> print(repr(led))
<LED pin=12 color='red'>
```

lurch, 2016-10-02:
I guess something like `rgb_led.color = red_led.color` might be interesting ;-)

> So I guess having the color value of any LED in the repr of the LED object would be nice.

Again IMHO that's only useful if you've named your LED object something generic like `led` rather than something more specific like `red_led` or `green`.

thijstriemstra, 2016-10-02:
> Again IMHO that's only useful if you've named your LED object something generic like led rather than something more specific like red_led or green.

True, but this way you don't have to say `redPowerLed`; the color never has to be part of the var name.

lurch, 2016-10-03:
> the color never has to be part of the var name.

Of course it doesn't, it's just helpful. But then at the other extreme you could just use single-letter variables everywhere ;-)

Devil's advocate: What `color` would an InfraRed LED have? :smiling_imp:

thijstriemstra, 2016-10-03:
Hehe right. I just think the color is a useful piece of metadata for these classes. I've already solved it with this class so feel free to close it. If you want a PR I'll make one.

lurch, 2016-10-03:
> so feel free to close it.

Let's see what @bennuttall and @waveform80 think first :)

bennuttall, 2016-10-03:
Is it a sufficiently useful addition?
How common is it?
Could its purpose be achieved another way already?
If we implemented `led.color`, would we do the same for button?
What else might people want to add to devices? Might you describe each servo in a particular way?

So the bigger question is do we want user-defined metadata associated with device objects?

I think not.

Imagine `leds` was a list of LEDs of different colours (so no LED object names). In a world where `led.color` is implemented and defined:

```python
for led in leds:
    if led.color == 'red':
        led.on()
```

However, without `led.color`, you would simply use data structures to maintain this:

```python
red_leds = [...]

for led in red_leds:
    led.on()
```

or

```python
leds = {
    'red': [...],
    'green': [...],
}
```

or even just keep a list/dict/set of pin numbers of each colour.

lurch, 2016-10-03:
> So the bigger question is do we want user-defined metadata associated with device objects?

Completely off-the-wall suggestion, but what if we added a `user_data` dict to the base `Device` class, which people were free to do whatever they wanted with? (And which the GpioZero code would never touch)
So @thijstriemstra could do:
```python
RED = QColor(237, 33, 54)
GREEN = QColor(83, 237, 89)

led1 = LED(15)
led1.user_data['label'] = 'System Error'
led1.user_data['color'] = RED

led2 = LED(19)
led2.user_data['label'] = 'System OK'
led2.user_data['color'] = GREEN
```

No idea if that's *actually* a good idea or not though. As @bennuttall points out, you could easily 'invert' this and have:
```python
RED = QColor(237, 33, 54)
GREEN = QColor(83, 237, 89)

led1 = {
    'led': LED(15),
    'label: 'System Error',
    'color': RED
}

led2 = {
    'led': LED(19),
    'label': 'System OK',
    'color': GREEN
}
```

bennuttall, 2016-10-03:
It might be best to leave this out of the library. A way of adding it on the user side could be:

```python
class QLED(LED):
    def __init__(self, etc, color):
        self.color = color
        super().__init__(self, etc)
```

(note the etc should be whatever other args LED takes, and it would probably have to be a named arg)

lurch, 2016-10-03:
...or :
```python
class QLED(LED):
    def __init__(self, color, *args, **kwargs):
        self.color = color
        super().__init__(self, *args, **kwargs)
```

thijstriemstra, 2016-10-03:
Thanks for the feedback, closing.

closed 2016-10-03 by thijstriemstra