gpiozero/gpiozero #480: How to configure non-physical buttons to switch relay board

linuxson27, 2016-10-03:
Hi

Firstly, thank you for this awesome library! :)
I am busy with a simple PyQt5 application that contains two buttons which will switch relay channel 1 ON/OFF of my relay board which is connected to my RaspberryPi. Currently I am getting some weird errors, especially when I try and switch the relay off again. Apparently the GPIO pin is already in use?
My question is, how do I close/flush the GPIO pin? (see code below)
I am fairly new to this, so please excuse any errors I might have in my code

`# ===========
# IMPORTS
# ===========
from gpiozero import LED
import sys
from PyQt5 import QtCore, QtGui, QtWidgets
import relayONOFF_GUI

# ==================
# MAIN APP INIT
# ==================
class RelayApp(QtWidgets.QMainWindow, relayONOFF_GUI.Ui_win):

    def __init__(self, parent=None):
        super(RelayApp, self).__init__(parent)
        self.setupUi(self)

        # CONNECTIONS
        self.ON_btn.clicked.connect(self.switch_ON)
        self.OFF_btn.clicked.connect(self.switch_OFF)

    # ===========
    # CALLBACKS
    # ===========
    def switch_ON(self):
        print("Relay switched on")
        led = LED(17)
        self.ON_btn.when_pressed = led.on

    def switch_OFF(self):
        print("Relay switched off")
        led = LED(17)
        self.OFF_btn.when_pressed = led.off

app = QtWidgets.QApplication(sys.argv)
form = RelayApp()
form.show()
app.exec()`

Do I need to move the pin assignment outside my two functions and keep them in the class?
Any help with this would be MUCH appreciated!

Thank you

lurch, 2016-10-03:
I'm not familiar with PyQt, but from the GpioZero point of view you should only create the LED object once, and store it as a class attribute, and you shouldn't be reassigning the callbacks from the inside of other callbacks. I dunno why you're using both `clicked` and `when_pressed` callbacks, but you might want to try changing the code to something like:

```python
# ===========
# IMPORTS
# ===========
from gpiozero import LED
import sys
from PyQt5 import QtCore, QtGui, QtWidgets
import relayONOFF_GUI

# ==================
# MAIN APP INIT
# ==================
class RelayApp(QtWidgets.QMainWindow, relayONOFF_GUI.Ui_win):

    def __init__(self, parent=None):
        super(RelayApp, self).__init__(parent)
        self.setupUi(self)
        self.led = LED(17)

        # CONNECTIONS
        self.ON_btn.clicked.connect(self.switch_ON)
        self.OFF_btn.clicked.connect(self.switch_OFF)

# ===========
# CALLBACKS
# ===========
    def switch_ON(self):
        print("Relay switched on")
        self.led.on()

    def switch_OFF(self):
        print("Relay switched off")
        self.led.off()

app = QtWidgets.QApplication(sys.argv)
form = RelayApp()
form.show()
app.exec()
```

linuxson27, 2016-10-03:
Thank you lurch, I will give it a try. Not sure how I missed that on the button connections...oops

closed 2016-10-03 by bennuttall

linuxson27, 2016-10-03:
Just wanted to let you know it worked! Thanx once again for the help lurch. One question though...is there any way I can stop the relay from switching on as soon as the app is started? I basically want to check the state of the GPIO pins and then keep it from "actuating" until the button is clicked

lurch, 2016-10-03:
`initial_value` :-)
https://gpiozero.readthedocs.io/en/v1.3.1/api_output.html#led

linuxson27, 2016-10-03:
Ah! Thanx :)

I tried messing around with a bit, simple if statement got it sorted:

    class RelayApp(QtWidgets.QMainWindow, relayONOFF_GUI.Ui_win):
    
        def __init__(self, parent=None):
            super(RelayApp, self).__init__(parent)
            self.setupUi(self)
    
            self.led = LED(17)
    
            if self.led.on() == True:
                self.led.off()
            else:
                pass

lurch, 2016-10-03:
That doesn't do what you think... `on()` is a function with no return-value. To find out if the LED is __currently__ lit, you need to use https://gpiozero.readthedocs.io/en/v1.3.1/api_output.html#gpiozero.LED.is_lit

linuxson27, 2016-10-03:
Interesting....I tried using the is_lit method, but I can't seem to get access to it within the class? Also, my if statement seems to work perfectly, so I am not entirely sure why that is. I think it might have something to do with the fact that the pins only become active as soon as I start up the app, and isn't already currently active?

linuxson27, 2016-10-03:
And a few more things I have noticed...the on() and off() values seem to have opposite effects. I am currently using the off() call to switch on my relay base, and the on() call to switch it off. Also, when I print out the is_lit value on when it's switched on (which I assume is supposed to return True), it returns False instead, and vice versa

lurch, 2016-10-03:
> I tried using the is_lit method, but I can't seem to get access to it within the class?

Not sure I understand what you mean. Could you paste some example code?
You need to be aware that `is_lit` is a property rather than a method, so you'd do: `if led.is_lit: ` and **not** `if led.is_lit(): `.

> Also, my if statement seems to work perfectly, so I am not entirely sure why that is.

As I already mentioned, `on()` is a function which doesn't have any return value, so:
```python
        if self.led.on() == True:
            self.led.off()
        else:
            pass
```
is actually identical to:
```python
        self.led.on()
        if None == True:
            self.led.off()
        else:
            pass
```
which is identical to:
```python
        self.led.on()
        if False:
            self.led.off()
        else:
            pass
```
...which is (effectively) identical to:
```python
        self.led.on()
```
:-)

> And a few more things I have noticed...the on() and off() values seem to have opposite effects.

Try setting `active_high = False` ;-)
https://gpiozero.readthedocs.io/en/v1.3.1/api_output.html#led

linuxson27, 2016-10-04:
Wow, thank you for all the info, much appreciated!

linuxson27, 2016-10-04:
Ok, so to cap things off...this is what the finished code looks like (in case someone else needed this)
Be advised though that I used Qt5 to create the app window:

# ===========
    # IMPORTS
    # ===========
    from gpiozero import LED
    import sys
    from PyQt5 import QtCore, QtGui, QtWidgets
    import relayONOFF_GUI

    # ==================
    # MAIN APP INIT
    # ==================
    class RelayApp(QtWidgets.QMainWindow, relayONOFF_GUI.Ui_win):

        def __init__(self, parent=None):
            super(RelayApp, self).__init__(parent)
            self.setupUi(self)

            self.led = LED(17)
            self.led.active_high = False

            if self.led.is_lit == True:
                self.led.off()

            # CONNECTIONS
            self.ON_btn.clicked.connect(self.switch_ON)
            self.OFF_btn.clicked.connect(self.switch_OFF)

        # ===========
        # CALLBACKS
        # ===========
        def switch_ON(self):
            print("Relay switched on")
            self.led.on()

        def switch_OFF(self):
            print("Relay switched off")
            self.led.off()

    app = QtWidgets.QApplication(sys.argv)
    form = RelayApp()
    form.show()
    app.exec()

lurch, 2016-10-04:
All of this:
```python
        self.led = LED(17)
        self.led.active_high = False

        if self.led.is_lit == True:
            self.led.off()

```
can simply be replaced with:
```python
        self.led = LED(17, active_high=False, initial_value=False)
```
As a side-note, `if something == True:` is identical to `if something:`

And when you're happy that your code is working correctly (i.e. you no longer need the 'debug prints') then:
```python
        # CONNECTIONS
        self.ON_btn.clicked.connect(self.switch_ON)
        self.OFF_btn.clicked.connect(self.switch_OFF)

    # ===========
    # CALLBACKS
    # ===========
    def switch_ON(self):
        print("Relay switched on")
        self.led.on()

    def switch_OFF(self):
        print("Relay switched off")
        self.led.off()
```
could _potentially_ be replaced with:
```python
        # CONNECTIONS
        self.ON_btn.clicked.connect(self.led.on)
        self.OFF_btn.clicked.connect(self.led.off)
```

bennuttall, 2016-10-04:
@lurch keeps saying exactly what I'm about to say.

@linuxson27 can you please use backticks for your code blocks? Like [this](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code)