gpiozero/gpiozero #481: Robot classes don't allow pwm=False

bennuttall, 2016-10-04:
I'm trying to use MockPin to test a robot script but I can't disable PWM:

```python
>>> robot = Robot((1, 2), (3, 4), pwm=False)
TypeError: __init__() got an unexpected keyword argument 'pwm'
>>> robot = CamJamKitRobot(pwm=False)
TypeError: __init__() got an unexpected keyword argument 'pwm'
```

I can create motors with `pwm=False`, and I don't believe there's any reason I shouldn't be able to do the same for Robots.

lurch, 2016-10-04:
Yep, see also my comments in https://github.com/RPi-Distro/python-gpiozero/pull/131#issuecomment-186251847 , #219 , #315 & #276  etc. (and possibly elsewhere)

We can either keep inflating the args to `Robot`, or we can use the ideas in #219 to create a `MotorMixin` 'interface' (which the `PhaseEnable` motors in #386 would also implement, and possibly a ContinuousRotationServo could implement too), and then we could allow `Robot` to be constructed directly from two `MotorMixin` objects.

Unfortunately I'm really busy at the moment, so won't be able to put any development into this idea for a while.

lurch, 2016-10-04:
> I'm trying to use MockPin to test a robot script but I can't disable PWM

Can you change your test script to use MockPWMPin ?

commit 26bfcbea (2018-09-21): Merge pull request #678 from RPi-Distro/nopwm-robot

Allow Robot classes to take pwm argument, close #481
commit beb5d3be (2018-09-21): Allow Robots and derivatives to take a pwm argument on init (and allow non-PWM use of Robots), close #481
closed 2018-09-21 by bennuttall