gpiozero/gpiozero #483: Cannot create AngularServo setting only one of min/max angle

thagrol, 2016-10-09:
Attempting to create a gpiozero.AngularServo object and supplying a non-zero value for one of min_angle or max_angle without supplying the other results in "OutputDeviceBadValue: Servo value must be between -1 and 1, or None"
e.g.:
`s=gpiozero.AngularServo(21, min_angle=-45)` errors
`s=gpiozero.AngularServo(21, min_angle=0)` does not

lurch, 2016-10-20:
I can't reproduce this using the examples you gave, but I **can** reproduce it using:
```python
s=gpiozero.AngularServo(21, min_angle=45)
```

However, if we now look at how this interacts with the rest of the [default-args](http://gpiozero.readthedocs.io/en/latest/api_output.html#gpiozero.AngularServo), we end up with:
```
initial_angle=0
min_angle=45
max_angle=90
```
so the error message given is correct (ignoring #477 for now) because initial_angle is now invalid.

So if you're restricting the range of the AngularServo's angles to exclude the default initial_angle of 0, then you obviously need to provide an initial_angle that _is_ in the correct range.

thagrol, 2016-10-20:
Agreed. And rechecked on my setup behaviour is as you found. Sorry.

lurch, 2016-10-20:
No problem, glad we found the cause.

closed 2016-10-20 by lurch