gpiozero/gpiozero #493: Multi Digit 7 Segment Displays

martinohanlon, 2016-10-23:
Multi digit 7 segment displays:

PR #488 for single digit 7 segment displays seems to be nearing completion (hopefully) so can I get some comments on how we might implement a multi digit 7 segment display.

They typically have:
 - 7 pins to turn on the led's
 - a common (anode or cathode) pin for each digit, a respective high or low will cause the led's to turn on for that digit
 - between 2 & 4 digits
 - an (optional) 8th led pin for a decimal point for each digit

Plexing is used to display different values on different digits.

There is also a 4 digit clock variant which has a 9th led pin for the colon in between digits 2 & 3, sometimes they also have decimal points after each digit as well - it feels like a seperate class for ClockDisplay would be a good idea tho.

My current thinking is there are 2 options:

Option 1
 - class MultiDigitSevenSegmentDisplay
 - extending SevenSegmentDisplay
 - Constructor would require:
   - tuple of led pins (in order a,b,c,d,e,f,g,dp)
   - tuple of digit pins (anode / cathode defined by the negative of active_high)
   - anything else supported by SevenSegmentDisplay
 - value would be ((leds), (digits))
   - an alternative pattern would be to abstract the complexity and show ((leds), (leds), (leds), (leds))
   - this would be 'easier' to consume but would result is less low level control
 - display method, def display(string, left_justify=True, refresh=0.0x)
   - would display the string, any full stops would be turned into DP.
   - display would need to kick off a thread in order to plex the display

Option 2
 - class MultiDigitSevenSegmentDisplay
 - extending CompositionDevice
 - Constructor would require:
   - tuple of led pins (in order a,b,c,d,e,f,g,dp)
   - tuple of digit pins (anode / cathode defined by the negative of active_high)
   - anything else supported by SevenSegmentDisplay
 - it creates multiple SevenSegmentDisplays
   - they would be on the same pins (is this possible?)
   - a digit_on/off() method would be required 
 - display method, def display(string, left_justify=True, refresh=0.0x)
   - would update the SevenSgement display values by plexing the display on each seven seg display and the digit pins

I favour option 1, multi seven segment displays are basically 1 seven segment display with multiple grounds. 

Thoughts?

SrMouraSilva, 2016-10-23:
In #357 exists discurtions for Multiplexed displays too.
I am referencing now for future analysis.

SrMouraSilva, 2016-10-23:
I suggest that:

* it also has an float_display method.
```def float_display(self, number, formatter='{:03.2f}'):```
https://github.com/RPi-Distro/python-gpiozero/issues/357#issuecomment-254584733
* `justify` is a attribute, not only a parameter in display method

> they would be on the same pins (is this possible?)

Yes, https://github.com/RPi-Distro/python-gpiozero/blob/f607a27c79404657f777687165ec4f6e2f205a11/gpiozero/spi.py#L363

martinohanlon, 2016-10-25:
@lurch @waveform80 got any views on this? particularly implementation options.

lurch, 2016-10-25:
> got any views on this?

I will do, when I'm less busy... (perhaps next week)
The suggestion I gave to @SrMouraSilva in the other thread was similar to your Option1, but still inheriting from CompositeDevice.

martinohanlon, 2016-10-27:
First attempt, up for comment.

```
class MultiSevenSegmentDisplay(SevenSegmentDisplay):
    def __init__(self, led_pins, digit_pins, **kwargs):
        pwm = kwargs.pop('pwm', False)
        active_high = kwargs.pop('active_high', True)
        initial_value = kwargs.pop('initial_value', False)
        if kwargs:
            raise TypeError('unexpected keyword argument: %s' % kwargs.popitem()[0])
        if len(digit_pins) < 2:
            raise ValueError('a MultiSevenSegmentDisplay must have more than 1 digit')
        self.digits = []
        for digit_pin in digit_pins:
            self.digits.append(DigitalOutputDevice(digit_pin, active_high=not active_high))
        self._display_thread = None
        
        super(MultiSevenSegmentDisplay, self).__init__(*led_pins, pwm=pwm, active_high=active_high, initial_value=initial_value)
    
    # needs changing to cope with full stops, so that decimal points are used, if the display has them
    def display(self, string, align_left=True, refresh_delay=0.007):
        """
        0.007 was chosen by trial and error, short enough so there was no
        flicker, long enough to still be bright
        """
        if len(self.digits) < len(string):
            raise ValueError('there are not enough digits to display the string')

        self._display_thread = GPIOThread(
            target=self._display, args=(string, align_left, refresh_delay)
        )
        self._display_thread.start()
    
    def _display(self, string, align_left, refresh_delay):
        for i in cycle(range(0, len(string))):
            super(MultiSevenSegmentDisplay, self).display(string[i])
            digit = i if align_left else len(self.digits) - len(string) + i
            self.digits[digit].on()
            if self._display_thread.stopping.wait(refresh_delay):
                break
            self.digits[digit].off()
    
    def _stop_display(self):
        if self._display_thread:
            self._display_thread.stop()
            self._display_thread = None
        
    def off(self):
        self._stop_display()
        super(MultiSevenSegmentDisplay, self).off()
    
    @property
    def value(self):
        digits_value = []
        for digit in self.digits:
            digits_value.append(True)

        leds_value = []
        for led in self:
            leds_value.append(led.value)

        return((tuple(leds_value), tuple(digits_value)))
        
    @value.setter
    def value(self, value):
        self._stop_display()
        leds_value = value[0]
        digits_value = value[1]

        if len(digits_value) != len(self.digits):
            raise ValueError('expected %s digit values' % len(digits_value))

        for i in range(len(self.digits)):
            self.digits[i].value = digits_value[i]

        #LEDCollection could do with a ._write
        for i in range(len(leds_value)):
            self[i].value = leds_value[i]

```

SrMouraSilva, 2016-10-27:
Please, add python highcolor

* I preffer SevenSegmentsDisplayBoard()
* Line 11 and 12
```python
        for digit_pin in digit_pins:
            self.digits.append(DigitalOutputDevice(digit_pin, active_high=not active_high))
```
Works in this case?
```python
displays = MultiSevenSegmentDisplay(..., digit_pins=[DigitalOutputDevice(2), DigitalOutputDevice(3)])
```
* Pycharm suggests change `string` to other word
```python
def display(self, string, align_left=True, refresh_delay=0.007):
```
* Line 26. if the user only a display?
```python
        self._display_thread = GPIOThread(
            target=self._display, args=(string, align_left, refresh_delay)
        )
```
* Line 32. `0` is necessary?
```python
for i in cycle(range(0, len(string))):
```
* Line 34. You can change the align verification to cycle iterator range in line 32. IMHO is more legible.
```python
            digit = i if align_left else len(self.digits) - len(string) + i
```
* Lines 47 - 49. If MultiSevenSegmentDisplay extends SevenSegmentDisplay, is necessary the `for`? Same in off method
```python
        super(MultiSevenSegmentDisplay, self).on()
        for digit in self.digits:
            digit.on()
```
* In the above case:
```python
display = MultiSevenSegmentDisplay(...)
display.display(53)  # Why not?
display.off()
display.on()
# What the display will be shows?
# 53 or 88?
# I vote in 53
```
* Please, add `display_float()` support
* Line 61. all `digits_value` are ever True?
```python
digits_value.append(True)
```
* `value getter` appears to me not be consistent. If Multi7SegDisplay displays 53, the value returns sometimes 5, sometimes 3.

martinohanlon, 2016-10-28:
@SrMouraSilva thank you for the review, they are really useful.

A couple of comments:

`displays = MultiSevenSegmentDisplay(..., digit_pins=[DigitalOutputDevice(2), DigitalOutputDevice(3)])`

It wouldnt work in this case, but why should it? You could pass a pin object, rather than a pin number, which is consistent with the rest of gpiozero.

`.value` should return the value of the led & digit pins at the time of calling, not the string being shown on the device, and if the device is being updated by `.display()` it will show alternative values - i see this as being the same as `.value` on `LED` when `.blink()' is being called.

Consistent with gpiozero `.on()` should turn all the LEDS and digits on, `.off()` should turn all the LEDS and digits off.

`.display_float()` - my intention is to make `.display()` handle strings as well as numbers, using the decimal point LED if a full stop is included. I see no difference between 1.234 and a.bcd.

SrMouraSilva, 2016-10-30:
```python
displays = MultiSevenSegmentDisplay(..., digit_pins=[DigitalOutputDevice(2), DigitalOutputDevice(3)])
```

I'm sorry. The real example is like http://gpiozero.readthedocs.io/en/v1.3.1/api_pins.html#rpigpiopin:
```python
displays = MultiSevenSegmentDisplay(..., digit_pins=[RPiGPIOPin(2), RPiGPIOPin(3)])
```
But I think it works. Then let it go.

> .value should return the value of the led & digit pins at the time of calling, not the string being shown on the device, and if the device is being updated by .display() it will show alternative values - i see this as being the same as .value on LED when `.blink()' is being called.

So I think the blink (also) has the slightest strange behavior.

> Consistent with gpiozero .on() should turn all the LEDS and digits on, .off() should turn all the LEDS and digits off.

[If LEDcollection works in this mode... ok](http://gpiozero.readthedocs.io/en/v1.3.1/_modules/gpiozero/boards.html#CompositeOutputDevice.on) (but I still find strange about it)

> .display_float() - my intention is to make .display() handle strings as well as numbers, using the decimal point LED if a full stop is included. I see no difference between 1.234 and a.bcd.

I believe that should be a little careful with `float`: If I need displays 23.456 in 4 displays 7 seg, exists possibilities like `0023`, `  23`, `023.4`, `23.45` ... In https://github.com/RPi-Distro/python-gpiozero/issues/357#issuecomment-254584733 i suggest a formatter parameter (`def floatdisplay(self, number, formatter='{:03.2f}')`)

martinohanlon, 2016-11-07:
I have added multi seven segment display to #488 .

I have left display_float out. `.display()` caters for decimal points. It seems reasonable that displays are formatted prior to being passed.

bennuttall, 2021-03-14:
Done!

closed 2021-03-14 by bennuttall