gpiozero/gpiozero #498: Add calibration support to Robot / Motor classes

jsphpl, 2016-11-12:
In order to make a `Robot` or any of its subclasses actually go forward (instead of making a slight turn to the left or right) when we call the `forward()` method, we need a way of calibrating the motors. This means setting a ratio between the duty cycles for the two motors, by which they are adjusted in order to make the robot go straight.

I'd suggest implementing this on the Motor level. Each motor would have an `adjustment` attribute, containing a float between 0 and 1 by which the duty cycle is multiplied before passing it on to the next-lower level. This attribute would be settable through the constructor like so:
```python
motor = Motor(17, 18, adjust=0.5)  # max speed becomes half speed
```

The robot then could be either constructed with a tuple of adjustment factors for the left and right motors respectively:
```python
robot = Robot(left=(4, 14), right=(17, 18), adjust=(0.9, 1))  # tune down the left motor by 10%
```
… or with a single value between -1 and 1:
```python
robot = Robot(left=(4, 14), right=(17, 18), adjust=-0.1))  # same effect as the tuple example above
```

What do you think? Any hints or suggestions?

If it's likely to be merged, i'm willing to provide a PR for this.

bennuttall, 2016-11-15:
Hehe https://twitter.com/ben_nuttall/status/798495071058264064

Also see https://github.com/RPi-Distro/python-gpiozero/issues/499

dglaude, 2017-06-17:
I have exactly the same issue with one motor being a bit stronger than another.
I expected to find such a calibration parameter.
I can do it in my code, but I guess most robot have that "not going in straight line" issue.