gpiozero/gpiozero #500: import error on jessie lite

thagrol, 2016-11-18:
When attempting to import giozero on a fresh install of jessie-lite the following import error is reported:

  File "/usr/lib/python2.7/dist-packages/gpiozero/__init__.py", line 58, in <module>
    from .devices import (
  File "/usr/lib/python2.7/dist-packages/gpiozero/devices.py", line 18, in <module>
    import pkg_resources

Error can be resolved by `sudo apt install python-pip`

Frankly, I strongly suspect the problem is with jessie-lite rather than gpiozero however may I suggest that this dependency be documented.

SteveAmor, 2016-11-27:
Is this dependency not documented at the bottom of this page?

https://gpiozero.readthedocs.io/en/v1.3.1/notes.html

Although, I agree it's somewhat hidden.

thagrol, 2016-11-27:
Yeah it is. Dunno how I missed that. Sorry.

closed 2016-11-27 by thagrol