gpiozero/gpiozero #503: no such attribute _blink_thread

bennuttall, 2016-12-01:
I'm repeatedly hitting this issue when `close()` is called on devices:

```
Exception ignored in: <object repr() failed>
Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/gpiozero/devices.py", line 162, in __del__
    self.close()
  File "/usr/local/lib/python3.5/dist-packages/gpiozero/boards.py", line 299, in close
    self._stop_blink()
  File "/usr/local/lib/python3.5/dist-packages/gpiozero/boards.py", line 375, in _stop_blink
    if self._blink_thread:
  File "/usr/local/lib/python3.5/dist-packages/gpiozero/boards.py", line 711, in __getattr__
    return super(TrafficLights, self).__getattr__(name)
  File "/usr/local/lib/python3.5/dist-packages/gpiozero/devices.py", line 307, in __getattr__
    raise AttributeError("no such attribute %s" % name)
AttributeError: no such attribute _blink_thread
```

One way to reproduce it is:

```python
from gpiozero import *
import pytest

with pytest.raises(ValueError):
    TrafficLights()
```

lurch, 2016-12-01:
Does #494 fix that? :-)
If not I'll add this to my list of things to look at over the weekend.

bennuttall, 2016-12-01:
It doesn't seem to! :(

lurch, 2016-12-12:
Feel free to give #507 a test ;-)

PR #507 (2018-02-19): Make close() nethods more exception-safe
If an exception is thrown during init, it's possible for a close() method (or
one of the methods called by close) to try accessing an attribute that hasn't
been created yet, which would throw an AttributeError exception. This commit
fixes that by ensuring that the close() methods (and methods called by them)
always use hasattr() or getattr() first, before trying to access the
attributes directly.

Fixes #503 and supersedes #494
commit 73572a9d (2018-02-19): Make close() nethods more exception-safe

If an exception is thrown during init, it's possible for a close() method (or
one of the methods called by close) to try accessing an attribute that hasn't
been created yet, which would throw an AttributeError exception. This commit
fixes that by ensuring that the close() methods (and methods called by them)
always use hasattr() or getattr() first, before trying to access the
attributes directly.

Fixes #503 and supersedes #494
closed 2018-02-19 by waveform80