gpiozero/gpiozero #505: Problem with MotionSensor()

bootsmann, 2016-12-06:
I am using a PIR sensor and on my RPI is running a webserver. Following code works fine with RPi.GPIO lib. But I want to go away from RPi.GPIO and so I tryed gpiozero. But second code is not working. Why? 
is working:
```python
PIR_PIN = 20
@app.route('/info',  method='POST')
def info_post(db):
    alarm =  int(request.forms.get('checkbox'))
    db.execute('UPDATE alarm SET status=? WHERE id=1',
                       (alarm,))
    GPIO.setmode(GPIO.BCM)
    GPIO.setup(PIR_PIN, GPIO.IN)
    
    if alarm:
        print 'Start PIR-Sensor'
        GPIO.add_event_detect(PIR_PIN, GPIO.RISING, callback=get_motion)
    else:
        print 'Exit'
        GPIO.cleanup()
```
not working:
```python
@app.route('/gpio',  method='POST')
def set_gpio_status():
    pir_status =  int(request.forms.get('pir_status'))
    pir = MotionSensor(20)
    
    if pir_status:
        print('start')
        pir.when_motion = get_motion
    else:
        print('exit')
        pir.close()
```

bennuttall, 2016-12-06:
When you say "not working", is there an error, or does nothing happen?

Does `get_motion` do anything with the PIR sensor?

bootsmann, 2016-12-06:
nothing happen. ``get_motion()`` function included just a print statement ``print 'motion detected``.
A stand alone program like https://gpiozero.readthedocs.io/en/v1.3.1/recipes.html#motion-sensor works fine. I am using bootle as framework.
Thx

SteveAmor, 2016-12-09:
A fix for this is to add ```global pir``` to ```set_gpio_status```

I'll leave it to someone else to fully explain the scope issue.
```python
@app.route('/gpio',  method='POST')
def set_gpio_status():
    global pir
    pir_status =  int(request.forms.get('pir_status'))
    pir = MotionSensor(20)
```

bootsmann, 2016-12-09:
@SteveAmor
Set `pir` to global works. But now I can not clean up  / close the Pin.
`Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/bottle.py", line 862, in _handle
    return route.call(**args)
  File "/usr/local/lib/python2.7/dist-packages/bottle.py", line 1732, in wrapper
    rv = callback(*a, **ka)
  File "app.py", line 73, in set_gpio_status
    pir = MotionSensor(20)
  File "/usr/local/lib/python2.7/dist-packages/gpiozero/devices.py", line 135, in __call__
    self = super(GPIOMeta, cls).__call__(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/gpiozero/input_devices.py", line 403, in __init__
    queue_len=queue_len, sample_wait=1 / sample_rate, partial=partial
  File "/usr/local/lib/python2.7/dist-packages/gpiozero/input_devices.py", line 135, in __init__
    super(SmoothedInputDevice, self).__init__(pin, pull_up)
  File "/usr/local/lib/python2.7/dist-packages/gpiozero/mixins.py", line 161, in __init__
    super(EventsMixin, self).__init__(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/gpiozero/input_devices.py", line 37, in __init__
    super(InputDevice, self).__init__(pin)
  File "/usr/local/lib/python2.7/dist-packages/gpiozero/devices.py", line 386, in __init__
    'pin %r is already in use by another gpiozero object' % pin`

Thx for info about DigitalInputDevice

SteveAmor, 2016-12-09:
Indeed, setting ```pir``` to global is not the answer.
I'm checking through the code now.

lurch, 2016-12-11:
I think the problem might be because you're using `pir = MotionSensor(20)` inside the `set_gpio_status` function, so it's trying to create a new MotionSensor on GPIO 20 _every_ time you call the `set_gpio_status` function, which is why you're getting the "pin is already in use" error.

What happens if you simply move `pir = MotionSensor(20)` outside of the function? (i.e. create it just once at the global scope)

lurch, 2016-12-11:
...whereas _before_ you added the `global` keyword, `pir` was being created and then going out of scope again at the end of each `set_gpio_status` function, and in GpioZero when an object goes out of scope it automatically gets closed, which would mean that the `when_motion` event-handler gets cleared.

SteveAmor, 2016-12-11:
@lurch if you use the Button class instead of the MotionSensor, it works ok (i.e. there is no scope issue).

lurch, 2016-12-11:
I'm afraid I don't have any experience with the "bottle framework". Could you post your whole source code, or even better a self-contained test example that reproduces this issue?

bootsmann, 2016-12-12:
Following code is running with cherrypy `pip install cherrypy` and works. But only with RPi.GPIO. Just uncomment gpiozero code and comment RPi code lines to show the differences.
```python
from bottle import Bottle, static_file, run, template, CherryPyServer, request
from gpiozero import MotionSensor
from RPi import GPIO

PIR_PIN = 20
app = Bottle()

def motion_detected(PIR_PIN):
    print('Motion detected!')  

@app.route('/gpio')
def get_gpio_status():
    return template('''<!DOCTYPE html>
<html lang="de">
    <head>
        <meta http-equiv="content-type" content="text/html; charset=UTF-8">
        <meta charset="utf-8">
        <script src="https://ajax.googleapis.com/ajax/libs/jquery/1.12.4/jquery.min.js"></script>
        <script language="javascript" type="text/javascript">
            $(document).ready(function() {
            $('button').on('click', function(event) {
                event.preventDefault();
                $.ajax({
                    type: "POST",
                    cache: false,
                    url: $(this).parent('form').attr('action'),
                    data: $(this).parent('form').serialize() + '&' + $(this).attr('name') + '=' + $(this).val(),
                });
            });
        });
        </script> 
    <body>
        <nav class="navbar navbar-inverse navbar-fixed-top">
            <div class="container">
                <div class="navbar-header">
                    <button type="button" class="navbar-toggle collapsed" data-toggle="collapse" data-target="#navbar" aria-expanded="false" aria-controls="navbar">
                    <span class="sr-only">Toggle navigation</span>
                    <span class="icon-bar"></span>
                    <span class="icon-bar"></span>
                    <span class="icon-bar"></span>
                    </button>
                </div>
                <div id="navbar" class="navbar-collapse collapse">
                    <ul class="nav navbar-nav">
                        <li><a href="/index">Home</a></li>
                        <li><a href="/gpio">GPIO</a></li>
                        </li>
                    </ul>
                </div>
            </div>
        </nav>
        <div class="container">
            <div class="starter-template">
                <br>
                <h3>GPIO with gpiozero</h3>
                <br>
                <br>
                <form>
                    <button type="submit" id="pir_status" name="pir_status" class="btn btn-lg btn-success" value="1">PIR ON</button>
                    <button type="submit" id="pir_status" name="pir_status" class="btn btn-lg btn-danger" value="0">PIR OFF</button>
                </form>
            </div>
        </div>
    </body>
</html>''')

@app.route('/gpio',  method='POST')
def set_gpio_status():
    pir_status =  int(request.forms.get('pir_status'))
    GPIO.setmode(GPIO.BCM)
    GPIO.setup(PIR_PIN, GPIO.IN)
    #pir = MotionSensor(20)
    
    if pir_status:
        #pir.when_motion = motion_detected
        #print(pir)
        print('start')
        GPIO.add_event_detect(PIR_PIN, GPIO.RISING, callback=motion_detected)

    else:
        #pir.close()
        print('exit')
        GPIO.cleanup()
        

run(app, server=CherryPyServer, host='10.0.2.114', port=8080, reloader=True, debug=True)

```

SteveAmor, 2016-12-15:
Here is a solution that I have tested:

```python
#!/usr/bin/env python

from gpiozero import MotionSensor        
from signal import pause
from time import sleep

status_switch = True

def motion_detected():
    print('Motion detected!')

def set_gpio_status():
    global status_switch
    global pir # REQUIRED

    if status_switch:
        print('start')
        pir = MotionSensor(21) # NOTE LOCATION
        pir.when_motion = motion_detected
        print(pir)
        sleep(10) # trigger motion during this time to test callback

    else:
        print('exit')
        print(pir) # check if already closed
        pir.close()
        print(pir)

    print('Leaving set_gpio_status')

set_gpio_status() # create pir object
sleep(10) # trigger motion during this time to test callback
status_switch = False
set_gpio_status() # close pir object
pause()
```

Returns:
```
start
<gpiozero.MotionSensor object on pin=GPIO21, pull_up=False>
Motion detected!
Motion detected!
Motion detected!
Motion detected!
Motion detected!
Leaving set_gpio_status
Motion detected!
Motion detected!
Motion detected!
Motion detected!
Motion detected!
exit
<gpiozero.MotionSensor object on pin GPIO21, pull_up=False, is_active=False>
<gpiozero.MotionSensor object closed>
Leaving set_gpio_status  
```

lurch, 2017-02-10:
Would somebody mind testing if #536 does indeed solve the problem for them?

delciotorres, 2017-02-10:
I'm sorry but I don't think it solves. It just allows different types of PIRs to work with MotionSensor class.

commit aef0b517 (2018-02-19): Moved pull_up from super to init, close #505
closed 2018-02-19 by waveform80