gpiozero/gpiozero #506: OSC support

samaaron, 2016-12-08:
It would be lovely if GPIO Zero supported both the sending and receiving of OSC messages from external processes (either on the same machine or even other machines on the network).

Essentially it would expose core API functionality via an OSC endpoint so that other processes could access GPIO functionality via simple network messages. Additionally, it would be lovely if it was super easy to *send* OSC messages out from an event handler associated with GPIO activity. This would then allow any other language with an OSC library (pretty much all of them) to leverage the power of GPIO Zero. 

For more info on OSC see: http://opensoundcontrol.org

bennuttall, 2017-03-21:
@samaaron Is OSC supported now? In the latest/upcoming release? Can you point me in the direction of some documentation/examples and I'll have a play :)

@jrobinson-uk do you know where I can find some theremin code?

lurch, 2017-03-21:
Whoooo, https://twitter.com/Poly_Core combined with blinkenlights! :smile:

closed 2021-02-27 by bennuttall