gpiozero/gpiozero #508: ENER314-RT board support

mkilivan, 2016-12-12:
Does _Energenie_ support ENER314-RT board?

whaleygeek, 2016-12-13:
Correct, the pyenergenie library at https://github.com/whaleygeek/pyenergenie was written from the ground up to support the ENER314-RT board. 

This ENER314-RT board is a two-way radio, so you can use this with both the green-button write-only devices, and also the full MiHome range. This includes the MiHome Adaptor Plus which reports energy usage. The python library I have written supports all of the currently purchasable products from Energenie.

Note however that eTRV support in the python is ongoing, and there are some members of the community working through the finer details of that at the moment. But all other products including the PIR sensor, the door/window sensor, the 3kW relay, and the single and multi gang sockets (both control-only and control and monitor) are tested with the library.

mkilivan, 2016-12-13:
@bennuttall Yes, I meant GPIO Zero and was referring to https://gpiozero.readthedocs.io/en/v1.3.1/api_boards.html#energenie. Since Python energenie is deprecated I am wondering if GPIO Zero going to support ENER314-RT board? 

Thanks @whaleygeek, I used the pyenergenie library and it works, but I would like to control my switches over Homebridge. I installed Homebridge Raspberry Pi but I couldn't turn on/off switches,

bennuttall, 2016-12-13:
Currently, GPIO Zero only supports the basic on/off switches. We'd like to add more support in future, it just depends on how much work is required and our developers getting hold of the kit. From a quick look at David's repo it looks like there's a hell of a lot of stuff going on!

Currently, GPIO Zero has no dependencies - everything can be used without installing additional libraries. Sure, SPI works better if you have spidev installed, but it works well enough without it. And it even (just) works without a base GPIO library like RPi.GPIO or pigpio. So if adding support for a new device required third-party libraries or building C extensions, we'd have to think of an alternative. For example, the commonly used neopixel library requires a C extension. We'd like to add support for controlling neopixels without this extension, ideally. If that's not possible then perhaps we'd have to consider a gpiozero-extras or gpiozero-neopixels library to provide GPIO Zero -like interfaces to these devices.

whaleygeek, 2016-12-13:
pyenergenie has no dependencies that I am aware of - just download the zip, unzip it and it runs. 

It has a pre-built bit-banged SPI library embedded inside it that is bound using a python ctypes interface. I designed it that way specifically because I hate dependencies, and I hate people having to be connected to the internet before they can install software. I used ctypes because the python2/python3 C calling interface changed between 2/3 versions, and I didn't want a different library for different versions of Python. There have also been one or two community contributions that have proved that it works out of the box on both Python 2 and Python 3 with no modification. 

The licence for pyenergenie is MIT, so please feel free to rip it off and do what you want with it, if you think that will get it in front of a wider audience.

Energenie (the company) have been very supportive of this work and have provided me with a wide range of loan equipment, which enabled me to develop and test with every one of their products, which now work with this package.

closed 2018-02-24 by bennuttall