gpiozero/gpiozero #509: New Revision code?

lurch, 2016-12-16:
I see http://elinux.org/RPi_HardwareHistory has just been [updated](http://elinux.org/index.php?title=RPi_HardwareHistory&diff=next&oldid=424926) to add a new `900021` revision code for the 512MB Model A+  (which is weird, as the 512MB A+ I bought still has the `0015` revision code).

@bennuttall is this anything you know / can find out about? Does GpioZero need to be updated to suit?

Hmmm, https://www.google.co.uk/search?q=900021+raspberry+pi is interesting.

bennuttall, 2016-12-16:
@ghollingworth do you know why some new A+ would have this revision?

We also need to detect the new Japanese manufactured Pis. Gordon, do you know what the identifier for this is?

ghollingworth, 2016-12-16:
If you look here:

https://github.com/AndrewFromMelbourne/raspberry_pi_revision

Someone has nicely described the new scheme (I assume his name is Andrew and he comes from Melbourne!)

So the revision makes sense, A+ with 512M manufactured by Sony UK

lurch, 2018-12-24:
@bennuttall I guess this can be closed now?

closed 2019-01-15 by bennuttall