gpiozero/gpiozero #511: controlling ACT LED gives a warning message

TonyLHansen, 2017-01-01:
creating an LED with the Activity LED or Power LED GPIO generates an erroneous warning message:

{{
/usr/lib/python3/dist-packages/gpiozero/pins/rpigpio.py:95: PinNonPhysical: no physical pins exist for GPIO47
  'no physical pins exist for GPIO%d' % number))
/usr/lib/python3/dist-packages/gpiozero/pins/rpigpio.py:95: PinNonPhysical: no physical pins exist for GPIO35
  'no physical pins exist for GPIO%d' % number))
}}

On A and B, the status ACT LED is controlled by GPIO 16, active low. On a A+, B+ and beyond, the status ACT LED is controlled by GPIO47, and its actually controlling a FET, so on/off is inverted, with high being ON. In addition, on A+, B+ and beyond, you can also sense the status of the PWR LED, that is the brownout detector driving it, by using GPIO35 to read it out, but if you change GPIO35 to an output you can actually control the PWR LED now too, same way as the ACT LED.

Other than the warning message, using LED(47) and LED(35) controls the ACT and Power LEDs just fine on the newer RPIs, and LED(16) controls the ACT LED just fine on the older RPIs.

IMO, there should be no warning message when these are instantiated. Yes, there is a workaround for the warning message, as in:

{{
with warnings.catch_warnings():
    warnings.simplefilter("ignore")
    led = LED(blinkpin)
}}

but this code should not be necessary.

I have two suggestions: add some intelligence to gpiozero.pi_info to return the LED GPIO numbers for the ACT and Power LEDs (return None if not available). This allows portable code to be written for both types of RPIs. Then change the code to allow those pins to be used with LED() without generating the warning message.

lurch, 2017-01-01:
Yeah, see also https://gpiozero.readthedocs.io/en/v1.3.1/recipes.html#controlling-the-pi-s-own-leds and https://github.com/RPi-Distro/python-gpiozero/issues/148#issuecomment-181649790

lurch, 2017-01-02:
Given that the PWR and ACT leds are the only non-header GPIOs you're likely to want to be able to control, and given that it _only_ makes sense for these to be LED objects (i.e. it'd make no sense to construct a Button object on one of these GPIOs), I think a better approach might be to have the `PiPowerLED` and `PiActivityLED` classes that I mentioned in #148 (with `pwm` and `initial_value` being the only init-params), with those classes temporarily disabling and re-enabling the `PinNonPhysical` warning during init?

TonyLHansen, 2017-01-03:
One concern I have with disabling the warnings using
```
with warnings.catch_warnings():
    warnings.simplefilter("ignore")
    led = LED(blinkpin)
```
is: what other warnings that ARE important might be hidden by doing this? Are there any?

I'm reminded of gcc/g++'s -Wall -Wnowxyz options. Gpiozero is currently in the land of -Wall, that is, warn about everything. Another possibility would be give us the opportunity to shut off any given warning message. For example, if the `GPIOZeroError` class had a boolean property `printWarning = True`, the code could check if that is set before calling `warning.warn()`. We could then set `PinNonPhysical.printWarning = False` in our code when we don't want the warning. The other places that call `warning.warn()` could do likewise.

TonyLHansen, 2017-01-03:
I'm concerned about disabling all warning messages, whether it's in the code calling the `LED` constructor, or in the constructors of a new `PiPowerLED` class. Just because I don't want the `PinNonPhysical` warning doesn't necessarily mean that I don't want some of the other warnings.

Thinking further on this, I'm reminded of the gcc/g++ options `-Wwhatever`,  `-Wnowhatever` and  `-Wall`. The option `-Wwhatever` turns on the warning for _whatever_, `-Wnowhatever` makes sure that _whatever_ is not warned about, and `-Wall` turns on ALL warnings except for those that are subsequently turned off. The current nanny warnings are comparable to `-Wall`, but with no way to turn any of the warnings off.

A very simple change is to add a class variable to `GPIOZeroWarning`, called something like `printWarnings`. This class variable would be inherited by all of the warning subclasses, such as `PinNonPhysical`, and could be checked individually before calling the `warning.warn()` function. The user code could then set that class variable to `False` before invoking something in GPIOZero that caused the warning message to appear.

I've just tried this code out and it works very simply. The user code then becomes:

```
from gpiozero import Button, LED, PinNonPhysical
PinNonPhysical.printWarnings=False
actled = LED(actpin)
```

Et voila, no warning message.

I'll be pushing a PR shortly.

lurch, 2017-01-03:
I can see what you're getting at by talking about "nanny warnings", but GpioZero **is** aimed primarily at beginners ;-)

Following the pattern in https://github.com/RPi-Distro/python-gpiozero/issues/148#issuecomment-243492795 I wonder if you could make your disable-warnings a context handler, to make it easier to only temporarily disable specific warnings?
i.e. instead of doing:
```python
from gpiozero import Button, LED, PinNonPhysical
PinNonPhysical.printWarnings = False
actled = LED(actpin)
PinNonPhysical.printWarnings = True
```
you could do something like:
```python
from gpiozero import Button, LED, PinNonPhysical
with PinNonPhysical.disable_warning():
    actled = LED(actpin)
```

?

waveform80, 2018-02-19:
Python's [warning filter](https://docs.python.org/3/library/warnings.html) is quite configurable. It's trivial to disable PinNonPhysical without touching any other warnings:

```python
import warnings
from gpiozero import Button, LED, PinNonPhysical

# This disables the PinNonPhysical warning for the rest of the script
warnings.simplefilter('ignore', category=PinNonPhysical)

actled = LED(actled)
```

bennuttall, 2018-02-20:
I think this can be closed

closed 2018-02-24 by bennuttall