gpiozero/gpiozero #517: MockPin.when_changed not working for output pins

bennuttall, 2017-01-07:
I've been trying to get something to work and realised MockPin is not firing `when_changed` at all when the pin function is output:

```python
from gpiozero import LED
from gpiozero.pins.mock import MockPin
from time import sleep

led = LED(2)
led.blink()

pin = MockPin(2)
pin.when_changed = lambda: print("pin changed")

while True:
    print(pin.state)
    sleep(1)
```

@waveform80

lurch, 2019-02-11:
Isn't `when_changed` used for detecting external events, i.e. with input pins? Why are you trying to use it with an output pin?
For the sake of comparison, what does e.g. RPi.GPIO (instead of MockPin) do in this situation?

bennuttall, 2019-02-11:
I can't remember exactly but it was probably related to #445 graphical mock pin.

waybackrider, 2020-03-14:
This would be extremely helpful in testing/simulation scenarios!