gpiozero/gpiozero #518: python-gpiozero should depend on python-pkg-resources

m-reiter, 2017-01-08:
Importing Button from gpiozero fails if python-pkg-resources is not installed:

```
    from gpiozero import Button
  File "/usr/lib/python2.7/dist-packages/gpiozero/__init__.py", line 58, in <module>
    from .devices import (
  File "/usr/lib/python2.7/dist-packages/gpiozero/devices.py", line 18, in <module>
    import pkg_resources
ImportError: No module named pkg_resources
```

So I guess a dependency should probably be added.

broo0ose, 2017-02-05:
I'm getting this as well on python and python3

bennuttall, 2017-02-27:
You need to install `python-dev` and/or `python3-dev`:

```
sudo apt install python-dev python3-dev
```

lurch, 2017-03-01:
...or you can just install python-pkg-resources and/or python3-pkg-resources :
```bash
sudo apt-get install python-pkg-resources python3-pkg-resources
```

(see https://github.com/RPi-Distro/python-gpiozero/issues/538#issuecomment-283366919 )

commit b43aeb8f (2017-06-13): Fix #518

Apparently debian splits pkg_resources out from setuptools. Still, we
need pkg_resources as a runtime dependency, not just a build dependency
as the pin factory entry points now rely upon it.
closed 2017-06-13 by waveform80