gpiozero/gpiozero #524: usage of sysclass error

vsivanagulu, 2017-01-10:
i want use sys class in native.py can you please help how to use that sysclass

bennuttall, 2017-01-10:
Do you mean you want to use `NativePin` to control the GPIO pins? See http://gpiozero.readthedocs.io/en/v1.3.1/api_pins.html#nativepin

Either use `NativePin` objects like so:

```python
from gpiozero.pins.native import NativePin
from gpiozero import LED

led = LED(NativePin(12))
```

or set an environment variable when running your script or launching your IDE:

```python
# gpio.py
from gpiozero import LED
from signal import pause

led = LED(12)

led.blink()

pause()
```

```bash
$ GPIOZERO_PIN_FACTORY=NativePin python3 gpio.py
```

Note that `NativePin` is not too reliable.

vsivanagulu, 2017-01-10:
ok , but  how the sys class is used for blink an led.
i'm using like this

import gpiozero.pins.native import GPIOFS
LED=led(GPIOFS(22))           for accessing sysclass , is this coreect or else    
led.blink()
pause()
 and i try to understand that total 3 classes are there in native.py 
1)class GPIOFS(object):

    GPIO_PATH = '/sys/class/gpio'

2)GPIOMemory(object):
3)class NativePin(LocalPin):

how these classes are used and which one is default called think gpiomem is clling
what to do ,if i want to interact with sys class not reg,
tqs for abouve ans to the team,

bennuttall, 2017-01-12:
You can't use `SYSFS` as a pin class. See my comments in #525.

closed 2017-01-12 by bennuttall