gpiozero/gpiozero #525: error of using gpiozero lib for native pin

vsivanagulu, 2017-01-12:
first of all thanks for this lib creators , i have done my projects very simple and easy..
i am a fresher for raspi and gpiozero , i want to do experiments to this gpiozero and also i need to do work on nativepin,
for pi pin is successfull working ,
like from
 gpiozero import *
 led=LED(16)
led.blink
sleep() ..............etc now i want to check  native pin and sys/class   for programming 

PLEASE, help how to use sys class in programming i used below is it right or else,
from gpiozero.pins.native import GPIOFS
led=LED(GPIOFS(16))
led.blink()
pause() .....etc any modification please tell ,THANKYOU...........once again for this lib.

bennuttall, 2017-01-12:
Hi - I've added code block syntax to your comment - it helps make it readable. Please use them in future. See [markdown reference](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code) for further guidance.

The `GPIOFS` class is not currently used. You cannot create `LED` objects using that class: you can only pin classes (`RPiGPIOPin`, `PiGPIOPin`, `RPIOPin`, `NativePin`, `MockPin` or `MockPWMPin`). If you create a device object using `NativePin` it will use an experimental pure Python implementation of interaction with the GPIO pins using `/dev/gpiomem` or `/dev/mem`, like so:

```python
from gpiozero.pins.native import NativePin
from gpiozero import LED

led = LED(NativePin(12))
```

You can also use environment variables to set your default pin factory to `NativePin` (or simply uninstall `RPi.GPIO`) so that it automatically falls back to using `NativePin` without you having to specify.

Note this will **not** use `GPIOFS`. I believe this was created in an attempt to implement `NativePin` that was unsuccessful. Also note that `NativePin` is experimental and not supported. We cannot guarantee it will work well. We recommend using one of the other pin implementations. Have a look at their respective source code repositories and documentation if you want to understand how they work under the hood. I recommend pigpio ([docs](http://abyz.co.uk/rpi/pigpio/index.html) | [source](https://github.com/joan2937/pigpio)).

I don't understand your requirement to use `GPIOFS`. Can you elaborate?

bennuttall, 2017-01-12:
Correction: `GPIOFS` is intended to be used when we get around to developing `NativePin` further.

closed 2017-02-27 by bennuttall