gpiozero/gpiozero #526: Interest in adding TB6560 Stepper Motor Driver

GamalielMasters, 2017-01-12:
I had to interface with a TB6560 stepper motor driver board for a project and have written code (using but not integrated with gpiozero) for controlling it.  It would be fairly simple for me to create a device to integrate it if there is any interest.

bennuttall, 2017-01-12:
There's an issue for stepper motors at #144. Nothing's been done as yet. If you can explain how they work, and provide working code, ideally using gpiozero's base classes (like [this example](https://github.com/topshed/PiClubSushi/blob/master/17-StepperBETA.pdf)) that would help.

How common are these boards? Are there many that work the same? Are there many different types? Don't want to make too many assumptions.

GamalielMasters, 2017-01-12:
I think they are fairly common, I have found them available from several vendors.  The ones I have been looking at are designed for motors around 1.0- 3.5A such as the NEMA 17 and NEMA 21 type steppers used in 3d Printers and CNC machines. It seems like there are a number of designs based on different chips, but for the most part, they all connect to the microcontroller the same way, so could be covered by the same device class.

Unlike the design of the document linked from #144 which uses a Darlington array to directly energize the motor coils, these take care of the high current / high voltage motor drive circuits and coil energizing order (including micro stepping modes) internally.  All you need from the microcontroller is a direction and step signals, and an optional enable signal if you want to be able to turn the motor off.  Some chips also offer a low power sleep mode and overload detection if desired.

I'll see what I can pull together from what I have written.