gpiozero/gpiozero #527: Failed to parse pi revision

suhirotaka, 2017-01-15:
I got an error saying:
> gpiozero.exc.PinUnknownPi: unable to parse new-style revision "a32082"

My raspi is made-in-Japan Raspberry Pi 3 Model B released in November 2016.
Its revision is:
```
$ cat /proc/cpuinfo | tail -3
Hardware	: BCM2709
Revision	: a32082
Serial		: 00000000a31110c1
```

I should have bought UK model...

lurch, 2017-01-16:
I guess we simply need to add the new Japanese manufacturer (with code '3') to https://github.com/RPi-Distro/python-gpiozero/blob/master/gpiozero/pins/data.py#L862

bennuttall, 2017-01-16:
@suhirotaka Sorry about that - that's an issue we will deal with in the next release. We need to add the Japanese manufacturer to the data table as @lurch points out, and really we ought to make sure gpiozero can still be used on unknown revision codes.

lurch, 2017-01-16:
https://www.youtube.com/watch?v=UZEttmAdtm8 and https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=166391

I've just updated http://elinux.org/RPi_HardwareHistory#Board_Revision_History

lurch, 2017-01-22:
And there's now also the CM3 and CM3L with revision code `a020a0`.

bennuttall, 2017-01-23:
Thanks @lurch

vsivanagulu, 2017-01-23:
Is GPIOZERO really works on non pi boards, can u please tell me how to do on non pi boards ,upto now i have worked on pi boards, now i want try with non pi boards ,

THANKS in Advance 
regards siva

bennuttall, 2017-01-23:
You can run gpiozero code on a PC that remotely controls the GPIO pins on a Raspberry Pi.

gpiozero only supports Raspberry Pi GPIO, not other boards.

vsivanagulu, 2017-01-23:
ok Thanks for ur valueble reply, i cant understand the code flow how the cfuctions and classes are calling,also native.py and data.py for a simple led programe below,

from gpiozero import LED
from time import sleep
led = LED(17)
while True:
    led.on()
    sleep(1)
    led.off()
    sleep(1)

thanks in advance,@bennuttall

vsivanagulu, 2017-01-23:
i have learned so many things and more knowledge  with gpiozero thanks for this to all who are participated to develope gpiozero.

bennuttall, 2017-01-23:
1. Please use [code blocks](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code) when pasting code, it makes it readable and keeps it indented.
1. This is off-topic - please post another issue or use the [forums](https://www.raspberrypi.org/forums/) or [stack exchange](https://www.raspberrypi.org/forums/) and I'll address your questions there

Thanks

lurch, 2017-01-23:
If the code is calling `native.py` then it means you don't have any other [Pin implementations](https://gpiozero.readthedocs.io/en/v1.3.1/api_pins.html) installed yet. The recommendation is to install RPi.GPIO with `sudo apt-get install python-rpi.gpio python3-rpi.gpio`

bennuttall, 2017-02-08:
@suhirotaka this has been fixed in #531. Can you test it on your *Made in Japan* Pi?

The easiest way is to download the fixed version of data.py:

```bash
sudo wget http://rpf.io/datapy -O /usr/lib/python3/dist-packages/gpiozero/pins/data.py
```

And try using gpiozero with Python 3. Let us know if it works!

suhirotaka, 2017-02-08:
It works! I'm glad to use this awesome GPIO interface on my raspi.
Many thanks :)

It might be better http://rpf.io/datapy redirects to https://raw.githubusercontent.com/RPi-Distro/python-gpiozero/edd7eb256cd666b0ec8ae9f2af0c3c90f73b1a62/gpiozero/pins/data.py

bennuttall, 2017-02-08:
Gah - my mistake. I have updated.

Glad to hear it worked.

bennuttall, 2017-03-19:
Note: this has been fixed in the 1.3.2 release. You should be able to upgrade using:

```
sudo apt install python-gpiozero python3-gpiozero -y
```

(which will install the latest version)

closed 2018-02-24 by bennuttall