gpiozero/gpiozero #528: Add min/max values to PWMLED.pulse()

GilchristT, 2017-01-15:
Would it be practical to add a minimum PWM level to the Pulse function for a LED? Most LEDs will have a cut off point where they'll simply not light which takes away from the smoothness of the fade.

Loving this library BTW, every time I go to write my own function to do something I find it's already included :-)

bennuttall, 2017-01-16:
Possibly - I imagine it would belong at the object level rather than just within `pulse`. A minimum pulse-width matching a value of 0 should be possible.

lurch, 2017-01-16:
Need to be careful of API-bloat... ;-)
But yeah, I guess having `minimum_pwm_value` and `maximum_pwm_value` as optional init-params on `PWMOutputDevice` (the parent of `PWMLED`), with everything then scaled 0->1 within those limits, might work. I guess it just adds a small overhead on converting the 'external' PWM-value to/from the 'internal' (real) PWM-value.