gpiozero/gpiozero #529: Do not error for unknown Pi revisions

bennuttall, 2017-01-16:
We can't have gpiozero not work on new Pi models, new manufacturers or other unknown information. We need to ensure that when `pi_info` is unable to parse the revision, it deals with it properly, rather than erroring immediately.

I think this can mostly be done by using `.get()` lookups on dicts with default values set.

Re: #527

lurch, 2017-01-16:
I guess it could simply display 'Unknown' for any fields in the revision-code that it wasn't able to parse?

joan2937, 2017-01-16:
Will you also have the problem a bit lower down with RPi.GPIO and RPIO.GPIO?

I think pigpio will ignore the issue and assume the board has the standard 40 pin expansion header.

PR #530 (2017-06-13): Make revision parser more lenient
I initially supplied default values (using `.get`) for **all** the fields encoded in the revision, but then I got test failures in `test_pi_info` and `test_pi_info_other_types` (because it wasn't raising `PinUnknownPi` as expected).

And thinking about it, it probably *does* make sense to still throw an error if we can't decode the `model`, because the `model` determines which physical GPIO pins are available (a future Pi might use different GPIOs for different purposes, and in theory writing to a 'reserved' GPIO that we're not supposed to might cause the Pi t [... cut]
closed 2017-06-13 by waveform80