gpiozero/gpiozero #533: Add clamped parameter to internal devices

bennuttall, 2017-01-26:
Rather than:

```python
from gpiozero import LEDBarGraph, CPUTemperature
from gpiozero.tools import clamped

leds = LEDBarGraph(2, 3, 4, 5, 6 pwm=True)
cpu = CPUTemperature()

leds.source = clamped(cpu.values, 0, 1)
```

would it be nice to be able to:

```python
from gpiozero import LEDBarGraph, CPUTemperature

leds = LEDBarGraph(2, 3, 4, 5, 6 pwm=True)
cpu = CPUTemperature(clamped=True)

leds.source = cpu.values
```

?

lurch, 2017-01-26:
Personally, I don't think this suggestion makes sense... why assume that `clamped` always means 0->1? Why only support `clamped` and not also `booleanized`? etc. etc.

BTW the code [here](https://gpiozero.readthedocs.io/en/v1.3.1/api_other.html#cputemperature) doesn't use `clamped` - does it not work properly?

bennuttall, 2017-08-21:
It does work until the value goes outside the range, and you get an `OutputDeviceBadValue` exception. I think it kills the source thread at that point. But I don't think it's worth doing anything about it - as you say, why not for other similar cases? I'll drop it.

closed 2017-08-21 by bennuttall