gpiozero/gpiozero #534: Problem with MCP3001 reading

dmongin, 2017-01-30:
Ther class MCP3001 is not working properly: the value given by the value is almost random between 0 and 1. It is weird though because the MCP3008 is working like a charm.
I managed the correct reading with the following function:

spi = spidev.SpiDev()
spi.open(0,0)
spi.mode = 0

def read_mcp3001():
    reading = spi.readbytes(2) # just need to read the first two bytes. First byte reads [DC,DC,DC,B9,B8,B7,B6,B5], the second reads [B4,B3,B2,B1,B0,DC,DC,DC] (DC : don't care)
    hi = reading[0] & 0x1f # putting zero where thereisn't any data
    lo = reading[1] & 0xf8 
    vin = (hi<<5) + (lo>>3) # 
    vin = vin*1.0/1024

    return vin

Could you find a way to implement in your code, or make it compatible ?
Thanks in advance

Denis

lurch, 2017-01-30:
Just for reference, this is the current GpioZero implementation: https://github.com/RPi-Distro/python-gpiozero/blob/master/gpiozero/spi_devices.py#L344

If you haven't already, can you try testing the latest `master` version of GpioZero from this GitHub repo?

bennuttall, 2017-02-28:
Note: I've added backticks to your issue to make the code readable.