gpiozero/gpiozero #535: Incomplete source tarball

Natureshadow, 2017-02-06:
The source tarball released on PyPI is incomplete due to some important files missing in MANIFEST.in.

Most importantly, LICENCE.txt is missing, which is a violation of the license itself under point 1 of the 3-clause BSD license. The source tarball thus is alcking all means of determining what license the source is published under, as the trove classifeier "BSD License" is more than ambigious.

Please fix MANIFEST.in to build complete source tarballs in the future. Please also include the changelog and, if possible, pg-sign the upload to PyPI.

bennuttall, 2017-02-06:
Will do - thanks. I see you're working on submitting to debian (thanks so much!) - please let us know if you need anything more.

lurch, 2017-02-07:
Hah ;-)  #456

Natureshadow, 2017-02-07:
> Will do - thanks. I see you're working on submitting to debian (thanks so much!) - please let us know if you need anything more.

Thanks! That should be it, I think ;).

Debian stretch boots on Raspberry 2 and 3 verbatim now, and adding the additional parts that Raspbian carries as diff is a priority (at least for Debian to make it work on RPi, but I think Raspbian would benefit from minimising the diff as well).

bennuttall, 2017-02-07:
Great!

mirabilos, 2017-02-07:
I was asked by @Natureshadow to upload this to Debian, but the missing licence file is a blocker. I just talked with ftpmasters (to get an authoritative answer), and the consensus was that, since the code is released under the BSD licence, which clearly states that the licence must be preserved in copies, nobody (except likely the original holder) may distribute the tarball **at all**, i.e. it cannot even be uploaded into Debian non-free without this missing file.

This is _extremely_ easy to fix upstream by just releasing an 1.3.2 version which is identical save for the missing file (and adjusting version numbers and other metadata as needed).

bennuttall, 2017-02-08:
Is it possible to wait for the next feature release (1.4), in a month or so? Is this package likely to land in stretch, or have we missed the freeze?

Natureshadow, 2017-02-08:
The package will not land in stretch, but...

```
git checkout 1.3.1
echo "include LICENSE.txt" >>MANIFEST.in
sed -i s/1.3.1/1.3.2/ setup.py
git commit -am "Include licence in sdist."
python setup.py sdist upload --sign
git tag -s 1.3.2
git checkout master
git merge 1.3.2
git push --tags
```

...seems simple enough to do it now, also to prevent removal from PyPI and Raspbian, where gpiozero cannot be distributed either right now.

-nik

bennuttall, 2017-02-08:
Done!

closed 2017-02-08 by bennuttall

Natureshadow, 2017-02-08:
Great, thanks!

Don't forget to update master :).

bennuttall, 2017-02-08:
Whoops - forgot to push without `--tags`

Natureshadow, 2017-02-08:
Ah... my mistake ;)!