gpiozero/gpiozero #538: No way to install gpiozero on Jessie lite

Arthurin, 2017-02-26:
Hello,
I can't install gpiozero using sudo apt-get install python-gpiozero, the package is not located.
There is no alternative listed in the documentation and google can't help me neither. It's quite frustrating since gpiozero is designed to make physical computing more accessible :/
Can i get some help please?

bennuttall, 2017-02-27:
There's no reason you couldn't install it on Jessie lite the way you described. The gpiozero package exists in the Raspbian repo so it's possible to install the usual way. If you're getting errors, perhaps you're not online or your package list needs updating first. You'll also need to install `python-dev` and/or `python3-dev`:

```
sudo apt update
sudo apt install python-dev python3-dev python-gpiozero python3-gpiozero
```

Arthurin, 2017-02-27:
I just have this error "E: Unable to locate package python-gpiozero".
I dit apt-get update and checked if python-dev was also updated.
Can't we find a link to the direct file to install it?

bennuttall, 2017-02-27:
I don't know why you're having this issue if you're online and your package list is updated.

You can try installing with pip:

```
sudo apt install python-pip python3-pip
sudo pip install gpiozero
sudo pip3 install gpiozero
```

lurch, 2017-02-27:
I also don't know why this wouldn't be working - if I get time this evening I'll try to reproduce on a fresh install of Jessie Lite.
In the meantime, the `.deb` files themselves are at http://archive.raspberrypi.org/debian/pool/main/g/gpiozero/

Arthurin, 2017-02-27:
Thank you very much for your responses, it worked with 
```
sudo apt install python-pip python3-pip
sudo pip install gpiozero
```
Maybe it's because I'm on Volumio (https://volumio.org/discover/). They are based on Jessie Lite but they may have modified the source file?
`volumio@volumio:/home/FTP$ cat /etc/apt/sources.list
deb http://archive.raspbian.org/raspbian jessie main contrib non-free
deb-src http://archive.raspbian.org/raspbian jessie main contrib non-free
`

lurch, 2017-02-27:
Ahhh, looks like they're using upstream "pure Raspbian" from raspbian.org, which is actually different from the downstream "RPF Raspbian" from raspberrypi.org (which contains a bunch of extra packages and apt sources).

And if you've not already done so, you'll also want to make sure that you've got RPi.GPIO installed, otherwise you might find GpioZero is a bit slow.

Closing this issue.

closed 2017-02-27 by lurch

lurch, 2017-03-01:
I can confirm that
```bash
sudo apt-get install python-gpiozero python3-gpiozero
```
works just fine on the Jessie Lite downloaded from https://www.raspberrypi.org/downloads/raspbian/
:smiley:

lurch, 2017-03-01:
Doh! Spoke too soon... ;-)

You also need to
```bash
sudo apt-get install python-pkg-resources python3-pkg-resources
```
(see #518 )

bennuttall, 2017-03-01:
Also, you'll probably want to install a gpio pin library:

```
sudo apt install python-rpi.gpio python3-rpi.gpio
```

or:

```
sudo apt install python-pigpio python3-pigpio
```

We should incorporate this into #523

lurch, 2017-03-01:
> you'll probably want to install a gpio pin library

https://github.com/RPi-Distro/python-gpiozero/blob/master/debian/control already Recommends `python-rpi.gpio` so this install is handled automatically :-)

bennuttall, 2017-03-01:
But that's a recommend, not a requirement, right?

lurch, 2017-03-01:
Raspbian automatically installs Recommends packages. See previous discussions at https://github.com/RPi-Distro/python-gpiozero/issues/359#issuecomment-225455765 #388 and #389