gpiozero/gpiozero #539: Allow source tools to work with tuples

bennuttall, 2017-02-28:
I can use source/values (without any source tools) to set an LED to match a button:

```python
led = LED(2)
button = Button(3)

led.source = button.values
```

And I can used the `negated` to set an LED to always be the opposite of a button:

```python
led.source = negated(button.values)
```

I can use the first technique to pair an `LEDBoard` and a `ButtonBoard` of the same size:

```python
leds = LEDBoard(2, 3, 4, 5)
buttons = ButtonBoard(6, 7, 8, 9)

leds.source = buttons.values
```

But I can't use `negated` in the same way:

```python
leds = LEDBoard(2, 3, 4, 5)
buttons = ButtonBoard(6, 7, 8, 9)

leds.source = negated(buttons.values)
```

Instead, I have to iterate over the zip of the two collections and apply negated to each one in turn:

```python
for led, button in zip(leds, buttons):
    led.source = negated(button.values)
```

(or if I were @waveform80, set the source to some crazy tuple comprehension)

I suggest we either extend functions like `negated` to work with tuples as well as single values, or provide additional functions which do the same for collections, e.g. `negated_multi` (eugh terrible name). I'd much prefer the former :)

lurch, 2017-02-28:
Haha, (almost exact) duplicate of #365 ;-P

closed 2017-02-28 by lurch

bennuttall, 2017-02-28:
Ah!