gpiozero/gpiozero #540: IR proximity sensor

jimbo12free, 2017-03-01:
I am trying to use and IR proximity sensor for my robotics project. I have ordered one online and nowhere can I find anything on how to even begin coding it. I am maybe being a bit thick but how would I start using this to get similar readings to what I would using a distance sensor if even possible?

bennuttall, 2017-03-01:
Is it one of these: https://www.sparkfun.com/products/13959 ?

If so, you can use the `DistanceSensor` class.

See the wiring diagram and programming guide here: https://www.raspberrypi.org/learning/physical-computing-with-python/distance/

And see our documentation here: http://gpiozero.readthedocs.io/en/stable/api_input.html#distance-sensor-hc-sr04

If not, can you find out what it is and let us know.

jimbo12free, 2017-03-01:
hi, thanks for getting back to us so quick! it is one of these http://www.ebay.co.uk/itm/181919489211?_trksid=p2057872.m2749.l2648&ssPageName=STRK%3AMEBIDX%3AIT

bennuttall, 2017-03-01:
That looks like a line sensor. See http://gpiozero.readthedocs.io/en/latest/api_input.html#line-sensor-trct5000

jimbo12free, 2017-03-01:
I thought so too so I was looking at that. I may be wrong but my understanding was you could use it in the same way as a UDS? eg. it would emit an ir beam, reflect off an object and the receiver pick it up. you could then use s=d/t etc to calculate a distance

bennuttall, 2017-03-01:
Maybe - I'm not sure!

closed 2017-03-01 by bennuttall

lurch, 2017-03-01:
Based on the description in the ebay listing, it sounds like it's a binary-output device, i.e. you can only get a YES/NO answer to "is something in range of the sensor?" with the sensor-range being manually adjusted by the potentiometer.
I.e. the sensor-module itself is doing the measuring and calculation that GpioZero does for the HC-SR04 sensors. Which makes sense, as the speed of light is **much** faster than the speed of sound, so using GPIO on the Pi wouldn't be fast enough.

jimbo12free, 2017-03-01:
so can this device be used on a pi at all? could I use the basic when in
range when out of range sort of thing? all I want it to do is tell me when
there is an object in front of my robot within say 30cm -which matches the
range. the reason I am not using an HC-SR04 is that the surfaces around my
home seem to be muffling the signal and I am not always getting accurate
readings or the robot is jumpy. my plan is to use this sensor to detect the
object, then use the distance sensor attached to a servo to look left and
right and calculate which side has a greater distance.

lurch, 2017-03-01:
Yup, just use it as a `DigitalInputDevice` (or a `Button`, if you really want).

jimbo12free, 2017-03-01:
so would the "if button.is_on:" command work with that?

I would also like to say thanks for how helpful you guys have been, I've
been pulling hair out all day over this haha!

jimbo12free, 2017-03-01:
sorry, if button.is_pressed***

lurch, 2017-03-01:
Yup, that should work fine. Make sure you start off by working through the [Recipes](https://gpiozero.readthedocs.io/en/v1.3.1/recipes.html#button) to ensure you get the syntax correct :)