gpiozero/gpiozero #541: Raspberry Pi Zero W support

driesdepoorter, 2017-03-02:
I tested today the library with the new Raspberry Pi Zero W and I get this error: 

> Traceback (most recent call last):
  File "/usr/local/lib/python3.4/dist-packages/gpiozero/pins/data.py", line 654, in pi_info
    ) = PI_REVISIONS[revision]
KeyError: 9437377

bennuttall, 2017-03-02:
Thanks - this is a known issue. See [this comment](https://github.com/RPi-Distro/python-gpiozero/pull/531#issuecomment-283603593).

I intend to merge this PR and do a new release as soon as possible.

For now, the easiest way to get it working is:

```bash
git clone https://github.com/RPi-Distro/python-gpiozero
cd python-gpiozero
git checkout -b support_extra_models origin/support_extra_models
sudo python setup.py install
sudo python3 setup.py install
```

Then try again and it should work. Please let us know either way!

lurch, 2017-03-02:
See #531 - we're hoping to get a new version released soon :-)

bennuttall, 2017-03-02:
@lurch do my instructions look correct to you?

lurch, 2017-03-02:
I think `git checkout -b support_extra_models origin/support_extra_models` can be reduced to just `git checkout support_extra_models` (the tracking-branch gets created automatically by git).

And I've never actually tested `python setup.py install` myself!

@driesdepoorter If you're not desparate, it might be easiest to just wait until the `apt-get` package gets updated, rather than having multiple versions of GpioZero installed.

driesdepoorter, 2017-03-02:
Thank you for the quick fix @bennuttall

dglaude, 2017-03-02:
Maybe the code should be "fail-safe" and not crash if the Pi is not
detected or if the Pi revision is not detected.
Some sane default (on what pin are available) could be used in that case.

I mean, every new hardware release the problem seems to be the same.
Doing the change in advance would be a spoiler the foundation might not
want...
Doing it after might be difficult depending on Debian schedule and other.

Regards

lurch, 2017-03-02:
@dglaude see #530 ;-)

closed 2017-03-02 by driesdepoorter

bennuttall, 2017-03-03:
A fix has been released. You'll find it on [PyPI](https://pypi.python.org/pypi/gpiozero), and it'll make its way into Raspbian shortly.

To update now:

```bash
sudo pip install gpiozero
sudo pip3 install gpiozero
```

But note, if you later update with `apt` your `pip` version will take precedence.

bennuttall, 2017-03-03:
Updated previous comment