gpiozero/gpiozero #542: TrafficLight object broken on Pi Zero W ?

aoakley, 2017-03-02:
It seems the TrafficLight object (and possibly LED in general?) is broken under the Pi Zero W. Tried two ZeroWs (aren't I lucky) and same error on both. Tried on a vanilla Zero with exactly same SD card and worked fine. Sadly prepping for #PiParty so not much time to analyse.

Note that my Python program uses two TrafficLights (simulates roadworks).

Headline error:

	Traceback (most recent call last):
	File "/usr/lib/python3/dist-packages/gpiozero/pins/data.py", line 654, in pi_info
    ) = PI_REVISIONS[revision]
	KeyError: 9437377

This is on a Raspberry Pi Zero with apt update && apt upgrade as of Thu 2 Mar 2017 approx 21:30pm

	uname -a
	Linux traffic0 4.4.48+ #964 Mon Feb 13 16:50:25 GMT 2017 armv6l GNU/Linux

Full error message:

	Traceback (most recent call last):
  	File "/usr/lib/python3/dist-packages/gpiozero/pins/data.py", line 654, in pi_info
    	) = PI_REVISIONS[revision]
	KeyError: 9437377

	Traceback (most recent call last):
  	File "/usr/lib/python3/dist-packages/gpiozero/pins/data.py", line 654, in pi_info
    	) = PI_REVISIONS[revision]
	KeyError: 9437377
	
	During handling of the above exception, another exception occurred:
	
	Traceback (most recent call last):
  	File "/usr/lib/python3/dist-packages/gpiozero/pins/data.py", line 517, in _parse_pi_revision
    	}[(revision & 0xff0) >> 4]
	KeyError: 12
	
	During handling of the above exception, another exception occurred:
	
	Traceback (most recent call last):
  	File "traffic2.py", line 4, in <module>
    	traffic1=TrafficLights(17,27,22)
  	File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 135, in __call__
    	self = super(GPIOMeta, cls).__call__(*args, **kwargs)
  	File "/usr/lib/python3/dist-packages/gpiozero/boards.py", line 704, in __init__
    	**devices)
  	File "/usr/lib/python3/dist-packages/gpiozero/boards.py", line 296, in __init__
    	super(LEDBoard, self).__init__(*args, **kwargs)
  	File "/usr/lib/python3/dist-packages/gpiozero/boards.py", line 226, in __init__
    	for name, pin_or_collection in kwargs.items()
  	File "/usr/lib/python3/dist-packages/gpiozero/boards.py", line 226, in <dictcomp>
    	for name, pin_or_collection in kwargs.items()
  	File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 135, in __call__
    	self = super(GPIOMeta, cls).__call__(*args, **kwargs)
  	File "/usr/lib/python3/dist-packages/gpiozero/output_devices.py", line 131, in __init__
    	super(DigitalOutputDevice, self).__init__(pin, active_high, initial_value)
  	File "/usr/lib/python3/dist-packages/gpiozero/output_devices.py", line 41, in __init__
    	super(OutputDevice, self).__init__(pin)
  	File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 69, in __init__
    	super(SourceMixin, self).__init__(*args, **kwargs)
  	File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 382, in __init__
    	pin = pin_factory(pin)
  	File "/usr/lib/python3/dist-packages/gpiozero/pins/rpigpio.py", line 85, in __new__
    	cls.PI_INFO = pi_info()
  	File "/usr/lib/python3/dist-packages/gpiozero/pins/data.py", line 625, in pi_info
    	result = pin_factory.pi_info()
  	File "/usr/lib/python3/dist-packages/gpiozero/pins/__init__.py", line 293, in pi_info
    	return pi_info(cls._PI_REVISION)
  	File "/usr/lib/python3/dist-packages/gpiozero/pins/data.py", line 671, in pi_info
    	) = _parse_pi_revision(revision)
  	File "/usr/lib/python3/dist-packages/gpiozero/pins/data.py", line 588, in _parse_pi_revision
    	raise PinUnknownPi('unable to parse new-style revision "%x"' % revision)
	gpiozero.exc.PinUnknownPi: unable to parse new-style revision "9000c1"
	Exception ignored in: <bound method LED.__del__ of <gpiozero.LED object closed>>
	Traceback (most recent call last):
  	File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 162, in __del__
    	self.close()
  	File "/usr/lib/python3/dist-packages/gpiozero/output_devices.py", line 144, in close
    	self._stop_blink()
  	File "/usr/lib/python3/dist-packages/gpiozero/output_devices.py", line 184, in _stop_blink
    	if self._controller:
	AttributeError: 'LED' object has no attribute '_controller'
	
Program listing:

	from gpiozero import TrafficLights 
	from time import sleep 
 
	traffic1=TrafficLights(17,27,22) 
	traffic2=TrafficLights(18,23,24) 
 	
	while True: 
	
  	traffic1.red.on() 
  	traffic2.green.on() 
	
  	sleep(4) 
	
  	traffic1.yellow.on() 
	
  	traffic2.green.off() 
  	traffic2.yellow.on() 
	
  	sleep(1) 
	
  	traffic1.red.off() 
  	traffic1.yellow.off() 
  	traffic1.green.on() 
	
  	traffic2.yellow.off() 
  	traffic2.red.on() 
	
  	sleep(4) 
	
  	traffic1.green.off() 
  	traffic1.yellow.on() 
	
  	traffic2.yellow.on() 
	
  	sleep(1) 
	
  	traffic1.yellow.off() 
	
  	traffic2.red.off() 
	traffic2.yellow.off()

aoakley, 2017-03-02:
Dupe of #541

closed 2017-03-02 by aoakley

bennuttall, 2017-03-03:
A fix has been released. You'll find it on [PyPI](https://pypi.python.org/pypi/gpiozero), and it'll make its way into Raspbian shortly.

To update now:

```bash
sudo pip install gpiozero
sudo pip3 install gpiozero
```

But note, if you later update with `apt` your `pip` version will take precedence.

bennuttall, 2017-03-03:
Updated previous comment