gpiozero/gpiozero #543: Pi Zero W not supported

pltarun, 2017-03-03:
Doing anything on a Pi Zero W kicks up this error:
```
Traceback (most recent call last):
  File "/usr/local/lib/python3.4/dist-packages/gpiozero/pins/data.py", line 654, in pi_info
    ) = PI_REVISIONS[revision]
KeyError: 9437377

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.4/dist-packages/gpiozero/pins/data.py", line 517, in _parse_pi_revision
    }[(revision & 0xff0) >> 4]
KeyError: 12

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.4/dist-packages/gpiozero/devices.py", line 135, in __call__
    self = super(GPIOMeta, cls).__call__(*args, **kwargs)
  File "/usr/local/lib/python3.4/dist-packages/gpiozero/output_devices.py", line 131, in __init__
    super(DigitalOutputDevice, self).__init__(pin, active_high, initial_value)
  File "/usr/local/lib/python3.4/dist-packages/gpiozero/output_devices.py", line 41, in __init__
    super(OutputDevice, self).__init__(pin)
  File "/usr/local/lib/python3.4/dist-packages/gpiozero/mixins.py", line 69, in __init__
    super(SourceMixin, self).__init__(*args, **kwargs)
  File "/usr/local/lib/python3.4/dist-packages/gpiozero/devices.py", line 382, in __init__
    pin = pin_factory(pin)
  File "/usr/local/lib/python3.4/dist-packages/gpiozero/pins/native.py", line 212, in __new__
    cls.PI_INFO = pi_info()
  File "/usr/local/lib/python3.4/dist-packages/gpiozero/pins/data.py", line 625, in pi_info
    result = pin_factory.pi_info()
  File "/usr/local/lib/python3.4/dist-packages/gpiozero/pins/__init__.py", line 293, in pi_info
    return pi_info(cls._PI_REVISION)
  File "/usr/local/lib/python3.4/dist-packages/gpiozero/pins/data.py", line 671, in pi_info
    ) = _parse_pi_revision(revision)
  File "/usr/local/lib/python3.4/dist-packages/gpiozero/pins/data.py", line 588, in _parse_pi_revision
    raise PinUnknownPi('unable to parse new-style revision "%x"' % revision)
gpiozero.exc.PinUnknownPi: unable to parse new-style revision "9000c1"
>>> exit()
Exception ignored in: <bound method LED.__del__ of <gpiozero.LED object closed>>
Traceback (most recent call last):
  File "/usr/local/lib/python3.4/dist-packages/gpiozero/devices.py", line 162, in __del__
```

I had to manually add 0x9000c1 to /usr/local/lib/python3.4/dist-packages/gpiozero/devices.py to fix this.

lurch, 2017-03-03:
Duplicate of #541

closed 2017-03-03 by lurch

bennuttall, 2017-03-03:
A fix has been released. You'll find it on PyPI, and it'll make its way into Raspbian shortly.

To update now:

```bash
sudo pip install gpiozero
sudo pip3 install gpiozero
```

But note, if you later update with `apt` your `pip` version will take precedence.

bennuttall, 2017-03-03:
Updated previous comment