gpiozero/gpiozero #550: Software debounce missing input events

thagrol, 2017-03-27:
Having created a gpiozero.Button object and set call backs for for when_held, when_pressed, and when_released if bounce_time is anything other than the default events are being randomly missed.

This occurs even when bounce time is set as low as 0.001

Code sees presses but not releases, releases without presses, holds without presses, holds without releases etc.

Hardware was a single tactile switch connected between GPIO 19 and ground. Pi Zero and a Pi2 showed the same behaviour.

lurch, 2017-03-28:
I don't have time to look into this right now, but I wonder if it might be another side-effect of the issues discussed in #385 ?

See also #378

DaddyTheRunner, 2017-06-11:
I was experimenting with the `bounce_time` parameter and a simple tactile switch and got some odd behavior.  Here's my code:

```
from gpiozero import RGBLED, Button
from time import sleep

rgb = RGBLED(25, 12, 16)
red_button = Button(6, bounce_time=1.0)

rgb.off()

red = (1,0,0)
green = (0,1,0)
delay = 0.5

## Signal ready to run
rgb.color = red
sleep(1)
rgb.color = green

## Testing button bounce
try:
    count = 0
    
    while True:
        red_button.wait_for_press()
        count += 1
        print(count)

except KeyboardInterrupt:
    # Clean up
    rgb.off()

```

When I ran the code with `bounce_time = None`, I got behavior I expected.  Typically a single momentary press would generate a stream of numbers between 1200 - 2500 and then stop when I released the button.

However, when I set `bounce_time` to any floating point value, the stream of numbers never stopped if I held the button down for less than the `bounce_time` until I interrupted the program with `ctrl-C`.  If I held the button down for a period of time greater than `bounce_time`, the stream of numbers stopped when I released the button.

Unless I don't understand how the `bounce_time` works (which is quite possible), it would appear that the `button.wait_for_press` function doesn't get "reset" until after the button has been held longer than `bounce_time` and then released.  That is not the behavior I would expect.

rdehuyss, 2019-06-11:
Unusable because of this...

bennuttall, 2019-06-11:
Can anyone compare gpiozero code with RPi.GPIO and show that it has a difference debounce?

gpiozero is using RPi.GPIO underneath (by default). If you change the pin factory you can test it using other libraries, like pigpio. https://gpiozero.readthedocs.io/en/stable/api_pins.html#changing-the-pin-factory

lurch, 2019-06-11:
@bennuttall I suspect this might be another side-effect of https://github.com/RPi-Distro/python-gpiozero/issues/697#issuecomment-463413405 i.e. RPi.GPIO missing rising/falling edge events?

thagrol, 2019-06-11:
Looking into this now but I'm hitting what appears to be a major issue with the NativeFactory on a Pi2: the entire Pi hangs when creating a Button object with no error message.

Will post a seperate issue with more details shortly.

thagrol, 2019-06-11:
OK, after some testing with alternate pin factories here are my findings:

RPIO: couldn't get the module to import
PiGPIO: appears to work correctly. A brief test did not exhibit this issue
Native: hung the entire Pi
Mock: not tested

I was using python 2 on a Pi 2 running raspbian stretch.

sailingbo, 2019-12-10:
Hi-  Just wanted to see if there have been any updates here.  I'm using a simple footpedal as button(17) and when bounce_time=0, it functions almost as expected.  All presses/releases/helds are accounted for.  However, occasionally, the footpedal will bounce and register 2 presses/releases when I only performed (or meant to perform) 1.  

As I understand it, preventing this is the whole point of the debounce function.  If I set button(17, bounce_time=0.1) or anything at all other than 0, the button becomes essentially non-functioning.   Only 1 out of every 10-15 presses/holds are registered.  I haven't found a correlation to the length of time I wait between presses/holds.  It just becomes unusable.

Any ideas?  Thanks!

bennuttall, 2019-12-10:
@sailingbo see my last comment https://github.com/gpiozero/gpiozero/issues/550#issuecomment-500743781 - come input on this would be helpful

closed 2021-03-12 by bennuttall