gpiozero/gpiozero #552: I have written classes for controling the digital potentiometer MCP41XXX. Anyone out there that can check that it is in line with your standards and merge it (if it is usefull)?

bmjbmj, 2017-03-30:
Classes for controlling this chip: http://ww1.microchip.com/downloads/en/DeviceDoc/11195c.pdf

--------------------

class DigitalpotensiometerDevice(SPIDevice):
 """
 Represents a digitalpotensiometer device connected to SPI (serial interface).
 """
 def __init__(self, **spi_args):
  super(DigitalpotensiometerDevice, self).__init__(shared=True, **spi_args)
#mode har

class MCP41xxx(DigitalpotensiometerDevice):
 """
 Extends :class:`DigitalpotensiometerDevice` to implement an interface for all $
 chips with a protocol similar to the Microchip MCP41XXX series of devices.
 MCP41010
 MCP41050
 MCP41100
 """
 def __init__(self, **spi_args):
        super(MCP41xxx, self).__init__( **spi_args)

 def set(self,value):
  #MCP41XXX protocol
  #
  #     Byte        0        1
  #     ==== ======== ========
  #     Tx   XXCCXXPP DDDDDDDD
  #
  #     P0 is bit 0
  #     P1 is bit 1
  #     C0 is bit 4
  #     C1 is bit 5
  #
  #     C1=0 C0=0 => NOP
  #     C1=0 C0=1 => Write data
  #     C1=1 C0=0 => Shutdown
  cmd=int('00010001', 2) #Write to pot zero
  data=self._spi.transfer([cmd,value])

-------------------
Use like this (for two MCP41100 chips):

Pot1=MCP41xxx(port=0, device=0)
Pot2=MCP41xxx(port=0, device=1)

while(True):
 Pot1.set(0)
 Pot2.set(0)
 print("0 ohm")
 sleep(10)

 Pot1.set(64)
 Pot2.set(64)
 print("25 kohm")
 sleep(10)

 Pot1.set(127)
 Pot2.set(127)
 print("50 kohm")
 sleep(10)

 Pot1.set(255)
 Pot2.set(255)
 print("100 kohm")
 sleep(10)

bennuttall, 2017-03-30:
Thanks! I've edited your subject and issue comment to display correctly - hope that's ok.

I'll see if I can get hold of one of these chips and give it a try. It'll need a few tweaks, but if it works very similar to existing MCPx chips, I don't see why we couldn't add it. It won't be in the next release, and might be waiting a while as we have such a large backlog right now - but we'll get to it when we can. Thanks again!

Frodox, 2017-09-23:
Bump!
@bennuttall , any news?:)

MountainMan370, 2023-11-11:
Can this be given a higher priority? I’m trying to have a stable code resource for the ICs I commonly use (e.g., ADC, RTC) and having a library for a digital potentiometer like the MCP4231 would be useful and probably to a lot of other users.

bennuttall, 2023-11-11:
Sorry no we're not actively working on adding new features to the library. Does the example usage above help you?

MountainMan370, 2023-11-11:
I’ll give the example a try. I just received 2- MCP41010. Thanks for
responding !