gpiozero/gpiozero #553: PiGPIOPin Servo errors on close()

bennuttall, 2017-04-02:
One small issue I've found that I think only exists in the #468 branch, but I've tested on the released version and determined it's just the PiGPIOPin factory: closing a Servo object (I've found this with Servo but nothing else) which has its source set *sometimes* raises exceptions. Two distinct examples:

```python
>>> from gpiozero import Servo, MCP3008
>>> from gpiozero.tools import scaled
>>> servo = Servo(4)
>>> pot = MCP3008()
>>> servo.source = scaled(pot.values, -1, 1)
>>> servo.close()
Exception in thread Thread-385:
Traceback (most recent call last):
  File "/usr/lib/python3.4/threading.py", line 920, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.4/threading.py", line 868, in run
    self._target(*self._args, **self._kwargs)
  File "/home/pi/python-gpiozero/gpiozero/mixins.py", line 80, in _copy_values
    self.value = v
  File "/home/pi/python-gpiozero/gpiozero/devices.py", line 313, in __setattr__
    return super(CompositeDevice, self).__setattr__(name, value)
  File "/home/pi/python-gpiozero/gpiozero/devices.py", line 159, in __setattr__
    return super(GPIOBase, self).__setattr__(name, value)
  File "/home/pi/python-gpiozero/gpiozero/output_devices.py", line 1062, in value
    self.pwm_device.pin.frequency = int(1 / self.frame_width)
AttributeError: 'NoneType' object has no attribute 'frequency'
```

and

```python
>>> from gpiozero import Servo
>>> from gpiozero.tools import sin_values
>>> servo = Servo(4)
>>> servo.source = sin_values()
>>> servo.close()
Exception in thread Thread-71:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gpiozero/pins/pigpiod.py", line 188, in _set_state
    if value != self._connection.get_PWM_dutycycle(self._number):
  File "/usr/lib/python3/dist-packages/pigpio.py", line 1188, in get_PWM_dutycycle
    return _u2i(_pigpio_command(self.sl, _PI_CMD_GDC, user_gpio, 0))
  File "/usr/lib/python3/dist-packages/pigpio.py", line 835, in _u2i
    raise error(error_text(v))
pigpio.error: 'GPIO is not in use for PWM'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.4/threading.py", line 920, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.4/threading.py", line 868, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 80, in _copy_values
    self.value = v
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 313, in __setattr__
    return super(CompositeDevice, self).__setattr__(name, value)
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 159, in __setattr__
    return super(GPIOBase, self).__setattr__(name, value)
  File "/usr/lib/python3/dist-packages/gpiozero/output_devices.py", line 1065, in value
    ((value - self._min_value) / self._value_range)
  File "/usr/lib/python3/dist-packages/gpiozero/pins/__init__.py", line 139, in <lambda>
    lambda self, value: self._set_state(value),
  File "/usr/lib/python3/dist-packages/gpiozero/pins/pigpiod.py", line 191, in _set_state
    raise PinInvalidState('invalid state "%s" for pin %r' % (value, self))
gpiozero.exc.PinInvalidState: invalid state "250" for pin GPIO4
```

joan2937, 2017-04-14:
Part of the issue may be how pigpio handles servos.

pigpio differentiates between generic PWM (GPIO in pigpio PWM mode) and servo PWM (GPIO in pigpio servo mode).

When a GPIO is in servo mode it has a fixed PWM frequency of 50 Hz and a pulse width attribute.  The pulse width attribute may be 0 (no pulses) or a number (in microseconds) between 500 and 2500.  The pulse width may be queried.  It has no duty cycle attribute.

An attempt to query the duty cycle of a GPIO in servo mode returns an error.  Servos are controlled by the pulse width.  For a given frequency you can map a duty cycle to a pulse width, but that is an aside.  All the documents will refer to the pulse widths.  For example an ESC may be driven at 500 Hz or 50 Hz, a pulse width of 1500 will always centre the servo, a dutycycle of 7.5% will not.

waveform80, 2017-06-16:
Ah, this is down to disabling the source in SourceMixin *after* closing the inherited device; there's probably a good reason for that ordering (probably to do with construction failure and handling the half-constructed object scenario) but evidently it needs changing / refining. I'll try and have a look this weekend now I've got the majority of the remote-SPI rebase done.

waveform80, 2017-06-19:
Hmm, as I suspected, tweaking the ordering of stuff in SourceMixin opened up a can of horrid object life-time worms. However, I think I've managed to stuff them all back in now!

For future reference, the resulting object life-time issue boiled down to the fact pins were keeping a weak reference to their factory which meant the factory could disappear *before* the pins at script shutdown. When the pins then went to do almost anything, stuff broke because of the missing factory reference. The reason for the weak reference was that there's a circular relationship here: the factory has a reference to the pins (to ensure a pin instance is a singleton for a given pin), but the pins have a reference to the factory so they can construct an address from the factory address and (in some cases like pigpio) so they can access things owned by the factory (like the pigpio connection).

It turns out it's best to simply use normal "strong" references here, and explicitly break the cycle on factory shutdown. This keeps everything alive as long as it needs to be, but does introduce the slight wrinkle that we need to explicitly close down devices on shutdown prior to dealing with the factory (left to its own, Python does the devices last as the atexit hook that shuts down the factory runs first). Thankfully, the reservations system (introduced in #468) is handily tracking all the devices we care about (i.e. all the ones with pins) so that wasn't too difficult to arrange.

Anyway, I haven't observed any of the issues above since making the above adjustments but given this was a Heisenbug I'm going to leave it running a test overnight to see if anything crops up. If it's all fine I'll poke the fix onto the end of #468.

closed 2017-06-20 by waveform80