gpiozero/gpiozero #557: pigpio PWM problem when duty cycle = 0.0

rgm3, 2017-04-22:
This causes a test failure in test_real_pins.py for duty cycle 0.0.  It seems that setting a pigpio PWM pin to state 0.0 leaves it floating / turns off its "output"-ness.

This gist demonstrates the issue:
https://gist.github.com/rgm3/b59842c7084595dd1ccd5673ea405e2b

I found this while trying to resolve issue #521 and running test_real_pins.py.
Weirdly, inserting any non-zero duty cycle before the "0.0" value causes the test to pass.

```python
for duty_cycle in (0.0, 0.1, 0.5, 1.0):
    # test fails for PiGPIOPins

for duty_cycle in(0.3, 0.0, 0.1, 0.5, 1.0):
    # now the test passes
```

bennuttall, 2017-04-23:
/cc @joan2937

joan2937, 2017-04-24:
Okay.  That is a fault within pigpio.

A change of PWM from dutycycle 0 to dutycycle 0 is in effect a NOP and the GPIO level is not initialised.  If it was last set high it will stay high.
```
2017-04-24 09:35:42 myGpioSetPwm: myGpioSetPwm 22 from 0 to 0
```

I will think about an appropriate fix.

For information.
```
$ pigs csi 5
$ cat /dev/pigerr
2017-04-24 09:35:42 gpioNotifyOpenInBand: fd=5
2017-04-24 09:35:42 gpioRead_Bits_0_31: 
2017-04-24 09:35:42 gpioHardwareRevision: 
2017-04-24 09:35:42 gpioSetMode: gpio=22 mode=0
2017-04-24 09:35:42 gpioSetPullUpDown: gpio=22 pud=0
2017-04-24 09:35:42 gpioGlitchFilter: gpio=22 steady=0
2017-04-24 09:35:42 gpioSetMode: gpio=27 mode=0
2017-04-24 09:35:42 gpioSetPullUpDown: gpio=27 pud=0
2017-04-24 09:35:42 gpioGlitchFilter: gpio=27 steady=0
2017-04-24 09:35:42 gpioSetMode: gpio=22 mode=1
2017-04-24 09:35:42 gpioSetPWMfrequency: gpio=22 frequency=100
2017-04-24 09:35:42 gpioSetPWMrange: gpio=22 range=10000
2017-04-24 09:35:42 gpioPWM: gpio=22 dutycycle=0
2017-04-24 09:35:42 myGpioSetPwm: myGpioSetPwm 22 from 0 to 0
2017-04-24 09:35:42 gpioGetPWMrange: gpio=22
2017-04-24 09:35:42 gpioGetPWMdutycycle: gpio=22
2017-04-24 09:35:44 gpioRead: gpio=27
2017-04-24 09:35:44 gpioNotifyClose: handle=0
```

joan2937, 2017-04-24:
Should now be fixed in pigpio V62.

closed 2017-04-25 by rgm3