gpiozero/gpiozero #559: Ignore false positives/negatives

ffleandro, 2017-05-10:
I've seen several forum threads where users experience some noise fluctuation in the GPIO input readings, for instance, without any physical change to a Button's state, the events when_released/when_pressed are being triggered.

This is normal and expected on "real world" scenarios, however I would like to know if there is a way to ignore events that happen too fast, for instance, a button is **active**, then a `release` and `press` event are triggered with a small interval in between, say 10ms for instance. I nee to ignore both the `press` and `release` rather only the `release` if I configure the `bounce_time`.

Is this possible with gpiozero? I've been trying to understand the library code, but I haven't found a way to do this without hacking too much.

The `Button.bounce_time´ property is not applicable, since that configures only the time to ignore events after a first event happened, which is not what I need. I need to ignore all events (including the first one) if another event happened really close to the first one.

ffleandro, 2017-05-11:
I discovered that `gpiozero` only registers for the `RPi.GPIO` interruptions, so a glitch filter needs to be handled by the `RPi.GPIO` before calling the callbacks `gpiozero` registers.

So without hacking too much on `gpiozero` it is not possible without the pin factory support for this kind of functionality.

Also I discovered that the only pin factory that supports this is `pigpio`: http://abyz.co.uk/rpi/pigpio/python.html#set_glitch_filter

closed 2017-05-11 by ffleandro

lurch, 2017-05-22:
Obviously any such "glitch filter" means that there's an added delay between the button being pressed, and your code being notified of the event. For devices like human-pressed buttons that's probably fine, but there'll likely be other scenarios in which it isn't fine.