gpiozero/gpiozero #560: Unable to use GPIO2 / GPIO27 'no physical pin exists'

mp05, 2017-05-14:
I have just picked up an old pi 1 model B and was trying the reaction time project in the MagPi April 2017 edition.

When I start the program I get the following messages:

/usr/lib/python2.7/dist-packages/gpiozero/pins/rpigpio.py:95: PinNonPhysical: no physical pins exist for GPIO27
  'no physical pins exist for GPIO%d' % number))
/usr/lib/python2.7/dist-packages/gpiozero/pins/rpigpio.py:95: PinNonPhysical: no physical pins exist for GPIO2
  'no physical pins exist for GPIO%d' % number))

If instead I alter the program to use GPIO7 / GPIO8 it works.

Is this a hardware fault or a support issue with GPIO2 / GPIO27 on the pi 1 model B?

bennuttall, 2017-05-14:
That's correct - GPIO27 doesn't exist on the original rev1 Pi. See [this comparison](http://www.raspberry-pi-geek.com/howto/GPIO-Pinout-Rasp-Pi-1-Rev1-and-Rev2) of the pinout. Looks like you can use any other pin except GPIO2 and 3 on the new layout. Note that the first 26 pins of the original Pi rev2 remained the same on all models thereafter, all the way to Pi 3 and Pi Zero.

gpiozero looks at which Pi model you're on and checks that pin exists on your Pi.

mp05, 2017-05-14:
Great thanks I was working from http://gpiozero.readthedocs.io/en/stable/recipes.html that shows that the first 26 pins are the same for all models.

Perhaps it should say except Pi 1 A/B Rev 1

closed 2017-05-14 by mp05

lurch, 2017-05-22:
I think we decided that the proportion of Rev 1 Model A and B Pis was so small, that it wasn't worth making the instructions more complicated for the vast majority of other users.
https://www.raspberrypi.org/blog/upcoming-board-revision/