gpiozero/gpiozero #561: Idea:  Shift Register Device

DaddyTheRunner, 2017-05-19:
I'm just in the idea generation phase and welcome any feedback and thoughts.  I think it might be a good idea to add a shift register device to GPIO Zero.

First, here's my rationale:
- The 74HC595 8 bit shift register is part of many Raspberry PI electronic starter kits
- Shift registers are an easy way to extend the number of digital output pins available
- Shift registers are a useful tool for controlling multiple LEDs and seven segment displays
- As best as I can tell from the documentation, creating a shift register device is not as simple as combining some of the currently available GPIO Zero devices

After reviewing the [GPIO Zero documentation](https://gpiozero.readthedocs.io/en/latest/index.html), I am struggling to figure out where exactly a shift register device should fit into the [GPIO Zero device hierarchy](https://gpiozero.readthedocs.io/en/latest/_images/device_hierarchy.svg).  From what I've studied so far, I think adding shift registers to the GPIO Zero library would involve a base shift register class and then sub classes that implement an interface to specific shift registers like the 595 series of shift registers.

However, where should the base shift register class fall?  Serial to parallel shift registers, like the 595 series, are purely an output device.  However, they are not single pin output devices.  Therefore, based on my limited understanding, it would not appear to make sense to have the shift register classes fall under GPIODevice or OutputDevice.  Likewise, CompositeDevice does not appear to be a good fit either since a shift register is a single function unit that is not composed of multiple devices.

I also don't believe a shift register is generic enough to be at the same level as CompositeDevice, GPIODevice, and SPIDevice.  Maybe what is needed is something like a MultiPinDevice at this level, from which you could build multi pin output devices and multi pin input devices.  The shift register related classes would then fit nicely under a MultiPinDevice.  BTW, the 555 timer IC is also commonly found in electronic starter kits and would be another candidate for MultiPinDevice.  But that's another idea for another day and another thread.

So back to my shift register device idea.  Here is an initial list of features I think such a device should support:
- setting/toggling the state of an individual register (think output pin) or group of registers
- sending a data stream to the shift register
- chaining multiple shift registers to create a larger shift register (the 595 IC supports this capability)
- In the case of the 595 shift register, use of the output enable (OE) pin to include PWM.  This is a simple way of adjusting the brightness of all LEDs attached to a shift register regardless of which ones are active.
- optional:  although the 595 IC is an output only device, the GPIO Zero library would have the information needed to report the values of the registers.

Like I said at the beginning, I am only in the idea stage and am interested in hearing what others think of my idea.  I would be interested in helping make this a reality; however, this would be a big step in my self guided learning of Python coding.  So if this idea goes on to the next stage, whatever that may be, I would definitely be looking for advice and support along the way.

lurch, 2017-05-20:
Interesting discussion...

> I also don't believe a shift register is generic enough to be at the same level as CompositeDevice, GPIODevice, and SPIDevice.

Not sure I'd agree - a GPIODevice indicates a device controlled by a single GPIO pin, a SPIDevice indicates a device controlled by a SPI interface (I expect at some point we may gain an I2CDevice class too), so IMHO a ShiftRegisterDevice _does_ belong at the same level. Not being that familiar with such devices myself, do they always work in the same fashion (with a clock and data-input pin), with the only difference being the number of output registers?

Hmm, I've just had a read of https://learn.sparkfun.com/tutorials/shift-registers and I guess we could have ShiftRegisterOuputDevice (SIPO) and ShiftRegisterInputDevice (PISO) as sub-classes of ShiftRegisterDevice?

> setting/toggling the state of an individual register (think output pin) or group of registers

AIUI wanting to change a single output, means re-sending all the output values at the same time?

> chaining multiple shift registers to create a larger shift register (the 595 IC supports this capability)

I guess that'd be easy to implement by just having a 'number_of_output_pins' parameter in the `__init__` method?

> optional: although the 595 IC is an output only device, the GPIO Zero library would have the information needed to report the values of the registers.

Yeah, I guess that'd be a pre-requisite if we wanted to change single output pins individually.

I'm not in the right mindset right now to contemplate if (or how) it would actually be possible, but given that GPIO Zero tries to focus on devices and objects, rather than merely "pins", it might be nice if ShiftRegisterOuputDevice also implemented an interface similar to CompositeOutputDevice, so that you could use a 74HC595 to construct e.g. an LEDBarGraph? And it should also allow source/values to work.

@waveform80 once mentioned that he'd like to add support for I2C and SPI IO-expanders to GPIO Zero (MCP23017, MCP23S17, MCP23S08, etc.), and AIUI the difference here is that IO-expanders can be used for both input and output, but shift-register devices are either output-only (SIPO) or input-only (PISO). So maybe that points towards ShiftRegisterOuputDevice actually using the PinFactory interface? (which I guess would gain you control of individual pins, and integration with the rest of the GPIO Zero classes, but would lose you the ability to set the value of all pins at once?)
Been a while since I've looked at GPIO Zero, so I've kind of forgotten how it all fits together... ;-)

DaddyTheRunner, 2017-05-20:
> Not being that familiar with such devices myself, do they always work in the same fashion (with a clock and data-input pin), with the only difference being the number of output registers?

Pretty much yes.  There are buffered ones, like the 74HC595, and unbuffered ones.  Since the 74HC595 is buffered it uses three input pins, one data pin and two clock pins.  It also has an output enable pin that can be used to implement PWM of all 8 output pins simultaneously.

> I guess we could have ShiftRegisterOuputDevice (SIPO) and ShiftRegisterInputDevice (PISO) as sub-classes of ShiftRegisterDevice?

I agree.  I was focusing on the output version because:
- It is easier.
- It is more common in starter kits
- I have some :smiley:

> AIUI wanting to change a single output, means re-sending all the output values at the same time?

You are correct, if you mean at the same time "in series".  I think this should be implemented only for shift registers that have a physical buffer like the 74HC595 because the chip's buffer enables the raspberry pi to change the state of specific pins.  With an unbuffered chip, all of the outputs would change multiple times as the data is serially sent to the shift register.  BTW, I'm not sure how common it is to find an unbuffered shift register.  So, IMHO it would be best to focus on the buffered variety.

The shift register object would have to internally store a copy of the last n bits sent to the physical shift register.  This same storage could be used to report the current values stored in the physical device.

> it might be nice if ShiftRegisterOuputDevice also implemented an interface similar to CompositeOutputDevice

> So maybe that points towards ShiftRegisterOuputDevice actually using the PinFactory interface?

I haven't looked at those interfaces yet, so I can't make an informed comment at this time.  But it's something I'll look at.  If it makes sense, is it possible to implement both interfaces?

lurch, 2017-05-20:
> With an unbuffered chip, all of the outputs would change multiple times as the data is serially sent to the shift register.

With unbuffered chips, maybe we could have a 'unsafe_mode` selectable from the constructor, that _would_ allow modification of individual outputs, with the caveat that it may lead to 'flickering' on the other outputs? That may be perfectly acceptable if it's only being used to drive e.g. a collection of LEDs?

> The shift register object would have to internally store a copy of the last n bits sent to the physical shift register. This same storage could be used to report the current values stored in the physical device.

Yeah, that perfectly fits the `.value` concept used within GPIO Zero :-)

> I haven't looked at those interfaces yet, so I can't make an informed comment at this time. But it's something I'll look at. If it makes sense, is it possible to implement both interfaces?

Dunno, that'd require me to go deep-diving into GPIO Zero again, and I'm afraid I'm going to be really busy until after next weekend at the earliest!
If you're very lucky @waveform80 might leave some comments in here, but he's been really quiet on GPIO Zero for a while too.

thiagohmoreira, 2019-04-15:
I'm sure you probably already have this at hand, but I've put together a simple implementation that works (for me ^^), and since there was no code in here... https://gist.github.com/thiagohmoreira/9bd6d234d93c93b705d4065a9111627a

It can deal with multiple 74HC595 chained, or other ICs with different number if bits/outputs. It don't support the OE line, but it's trivial to do so. 

I've based my decisions on what to descend from (CompositeDevice), code style/implementation (sorry I'm a very noob with Python, suggestions/critics are very welcome) based on the Motor class, as to me its a similar use case. We have a single IC that is controlled by multiple pins. I choose to use OutputDevice for the pins bacause from GPIODevice it would require some duplication.