gpiozero/gpiozero #565: MockPin and MockPWMPin documentation missing

bennuttall, 2017-06-30:
Any reason `MockPin` and `MockPWMPin` are left out of the [pins page](https://gpiozero.readthedocs.io/en/stable/api_pins.html) in the docs?

It's a shame these are not documented - we should ensure they are included. We should show people the API docs for these classes, and give the environment variables requires to use them by default - and state that they are cross-platform and ideal for testing code on a PC.

bennuttall, 2017-06-30:
@waveform80 perhaps we can nudge this into 1.4?

waveform80, 2017-07-14:
I've added a section to the #569 PR - what do you think? I can't add them directly into the table because that's also what defines the fallback order of pins (and mock isn't in that order, nor should it be), but I've added a section detailing them with a chunk of example code. I've also moved the "Utilities" section into its own chapter as it's rather important, but it was really getting buried at the end of a (now) very long chapter.

commit 978f8175 (2017-07-14): Merge pull request #569 from waveform80/pin-factory-docs

Fix #568, fix #572, fix #565
commit a0d78408 (2017-07-14): Fix #565

Add mock pins docs and tidy up some other bits of the pins docs
closed 2017-07-14 by waveform80