gpiozero/gpiozero #566: Can't use MockPWMPin as default pin factory

bennuttall, 2017-07-01:
Previously, it was possible to set the default pin factory to `MockPWMPin` with an environment variable:

```bash
GPIOZERO_PIN_FACTORY=MockPWMPin
```

Now as of v1.4 RC (current github master at this point in time), these have changed, I guessed (yes, [I had to guess](https://github.com/RPi-Distro/python-gpiozero/issues/565)) I could use:

```bash
GPIOZERO_PIN_FACTORY=mockpwm
```

But this is invalid. `mock` is valid, but that provides `MockPin` not `MockPWMPin`, so you can't use PWM:

```bash
$ GPIOZERO_PIN_FACTORY=mock python
>>> from gpiozero import PWMLED
>>> led = PWMLED(2)
PinPWMUnsupported...
```

However, after some digging around, I found a way around:

```bash
$ GPIOZERO_PIN_FACTORY=mock python
>>> import gpiozero
>>> from gpiozero import PWMLED
>>> from gpiozero.pins.mock import MockPWMPin
>>> gpiozero.Device.pin_factory.pin_class = MockPWMPin
>>> led = PWMLED(2)
>>> led.value
0.0
```

Obviously that's not setting the default pin factory on launch, but changing it on the fly.

It looks like pin factories have changed a lot under the hood (it's now about pin factories rather than pin classes - but pin factories seem to contain pin classes...), but it would be nice if it was possible to launch python with `MockPWMPin` as the default pin factory's pin class. Possible?

bennuttall, 2017-07-16:
@waveform80 did we say this was invalid, as there are many mock pin types now? And the new usage now requires you to explicitly set the pin class as per my example? Shall I close this?

waveform80, 2017-07-17:
Yeah, I'd say this was probably invalid now. There's too many mock pin classes to make this realistic and we provide a variety of ways of constructing different mock pin classes (from setting MockFactory.pin_class to the pin_class parameter in MockFactory.pin to the GPIOZERO_MOCK_PIN_CLASS environment var).