gpiozero/gpiozero #567: Button hold_time has no effect

bencaldwell, 2017-07-01:
Changes to the hold_time parameter of Button do not have any effect on delay between pin state change and callback triggering. I have used values from 2 to 300 for hold_time and all result in immediate callback of the shutdown() function. 

To reproduce just copy the shutdown recipe from the following link and try passing some different values for hold_time:
https://gpiozero.readthedocs.io/en/stable/recipes.html#shutdown-button.

My system:
* Pi Zero v1.3
* Raspbian GNU/Linux 8 (jessie)
* Linux 4.9.35+ armv6l
* python3-gpiozero stable 1.3.2

bennuttall, 2017-07-01:
I think there may be a misunderstanding here. Are you aware of both `btn.when_pressed` and `btn.when_held`?

The following code works for me:

```python
from gpiozero import Button

btn = Button(2, hold_time=5)
btn.when_pressed = lambda: print("pressed")
btn.when_held = lambda: print("held")
```

now press the button and hold it. The "pressed" alert should fire immediately, and the "held" alert should fire after 5 seconds.

Apologies if that's not the case - if you still experience this, please paste your code here. Thanks!

lurch, 2017-07-02:
Also, make sure you're using `shutdown_btn.when_held = shutdown` (without the brackets) and not `shutdown_btn.when_held = shutdown()` - the latter actually runs the `shutdown()` function immediately (instead of creating a reference to it), and would cause the Pi to shutdown as soon as the script is run.

bencaldwell, 2017-07-12:
I am using the call back correctly i.e. shutdown_btn.when_held = shutdown.
The issue is occurring because I have a "flickering" signal. This causes
the callback to fire before the hold time has elapsed. You can reproduce
this exactly by setting up a when_held callback with hold_time=10s. Flick
the assigned pin high and low rapidly and the callback will occur before
the 10s hold time has elapsed.

Ben Caldwell
+61 414 292 343
benny.caldwell@gmail.com

closed 2019-01-14 by bennuttall