gpiozero/gpiozero #568: RPIGPIOPin / RPiGPIOFactory confusion

bennuttall, 2017-07-02:
Previously, you could pass an `RPiGPIOPin` object as the pin reference to initialise a GPIO object like so:

```python
from gpiozero import LED
from gpiozero.pins.rpigpio import RPiGPIOPin

led = LED(RPiGPIOPin(12))
```

(example straight from the [1.3.2 docs](http://gpiozero.readthedocs.io/en/v1.3.2/api_pins.html#rpigpiopin))

However, that doesn't work on the current master. An [updated version of the docs](http://gpiozero.readthedocs.io/en/latest/api_pins.html#rpigpiopin) suggests the following invalid example:

```python
from gpiozero.pins.rpigpio import RPiGPIOFactory
from gpiozero import LED

led = LED(RPiGPIOPin(12))
```

Note the discrepancy: `RPiGPIOFactory` is imported but not used, and `RPiGPIOPin` is referenced but not imported.

I can produce an equivalent example using the following:

```python
from gpiozero.pins.rpigpio import RPiGPIOFactory
from gpiozero import LED

led = LED(RPiGPIOPin(RPiGPIOFactory(), 12))
```

Which is kinda yucky.

@waveform80 what's the recommended way of doing the original example above now?

lurch, 2017-07-03:
From having a look at the code, it definitely seems something is going wrong here, or not working as expected :-/

https://github.com/RPi-Distro/python-gpiozero/blob/master/gpiozero/pins/rpigpio.py#L43 itself also says `led = LED(RPiGPIOPin(12))` and yet https://github.com/RPi-Distro/python-gpiozero/blob/master/gpiozero/pins/rpigpio.py#L86 declares `RPiGPIOPin.__init__` as `def __init__(self, factory, number):`.
And https://github.com/RPi-Distro/python-gpiozero/blob/master/gpiozero/pins/rpigpio.py#L88 refers to `factory.pi_info` but https://github.com/RPi-Distro/python-gpiozero/blob/master/gpiozero/pins/rpigpio.py#L106 refers to `self.factory.pi_info` ?

IMHO it doesn't seem to make sense for RPiGPIOPin to have a `factory` parameter, so I wonder if:
```
    def __init__(self, factory, number):
        super(RPiGPIOPin, self).__init__(factory, number)
        self._pull = 'up' if factory.pi_info.pulled_up(self.address[-1]) else 'floating'
```
(inside `class RPiGPIOPin`) was _supposed_ to have been something like:
```
    def __init__(self, number):
        super(RPiGPIOPin, self).__init__(RPiGPIOFactory(), number)
        self._pull = 'up' if self.factory.pi_info.pulled_up(self.address[-1]) else 'floating'
```
? \*shrug\* (I haven't tried testing that, so I might be **totally** wrong!)

Whatever the resolution, I guess that this shows we need more thorough unit-tests? ;-)

waveform80, 2017-07-03:
Yup, my fault: forgot to update the docs. The code is all correct though - I'll push a PR for the docs in a mo. As to the discrepancy between `factory` and `self.factory` - that's just me being lazy. In the constructor they're the same thing (the super-call sets `self.factory = factory`) so I just skipped the `self.` but later on you can't as there's no such parameter - you can only get the factory from `self`.

bennuttall, 2017-07-03:
So I guess some of the new [remote gpio recipes](https://gpiozero.readthedocs.io/en/latest/recipes_remote_gpio.html) will need updating?

What's the usage for hot-swapping between the default and, say `RPiGPIOPin` or `PiGPIOPin` with a specific IP address?

waveform80, 2017-07-03:
There, pushed. Ahh... yes, remote gpio recipes will indeed need updating, I'll push another commit to the PR in a mo. Hopefully it should be obvious from what's there how to switch factories though (basically you just construct multiple factories and use them).

lurch, 2017-07-03:
I'm still not sure that it makes sense for RPiGPIOPin to have a factory parameter, but I guess that's the way the code is working now ;-)  (i.e. I've probably not understood all the details yet)

To prevent unpredictable results, should e.g. RPiGPIOPin's init check that `factory` is an instance of RPiGPIOFactory ? (and similarly for the other Pin classes)

> As to the discrepancy between factory and self.factory - that's just me being lazy.

Would you mind fixing that? From looking at just RPiGPIOPin's init method (without also looking at the super-call) it's not immediately obvious that `self.factory = factory` so IMHO some consistency would be nice, and avoid confusion ;-)

waveform80, 2017-07-03:
> I'm still not sure that it makes sense for RPiGPIOPin to have a factory parameter, but I guess that's the way the code is working now ;-) (i.e. I've probably not understood all the details yet)

By way of explanation for this oddity: it's not *really* necessary in the case of the RPi.GPIO implementation as there's no global "provider" that each pin needs to access in order to "do stuff". However, in the case of pigpio and native there is: for the pigpio implementation, the pins have to have access to the pigpio connection to do anything. The factory owns the connection, so the pins need a reference to the factory. With native, the pins need access to the `GPIOMemory` object (which the factory owns) to do anything, so it's the same situation. 

Rather than have differing constructors for the `Pin` descendants I figured it was "nicer" to have pins accept a mandatory `factory` param referring to their constructing factory. This is a cyclical reference, but that's actually okay as we want the factory to hang around as long as there's extant pins, and vice versa. Explicitly closing the factory (which happens in an `atexit` handler for the default pin factory) handles closing all the subordinate pins which breaks all the cyclic references. Depending on what people get up to with remote factories we *might* need to add some stuff to track extant factories and cleanly shut them *all* down at script termination but I'm not 100% sure that's required yet so I haven't looked into it too deeply.

> Would you mind fixing that? From looking at just RPiGPIOPin's init method (without also looking at the super-call) it's not immediately obvious that self.factory = factory so IMHO some consistency would be nice, and avoid confusion ;-)

Sure, I'll bung in a commit for that in a mo.

lurch, 2017-07-03:
And what about "To prevent unpredictable results, should e.g. `RPiGPIOPin's` init check that `factory` is an instance of `RPiGPIOFactory` ?" ;-)

Thanks for the explanation. I've not been python-ing for a while, but does it have anything like 'singleton objects' so that if `RPiGPIOPin` was passed a `None` factory, it'd get the singleton `RPiGPIOFactory` object? (which would create the factory, if it doesn't exist yet). I guess the only Pin implementation that might have different factory objects would be `PiGPIOFactory`, but even then it could just get the localhost singleton if given a `None` factory?
(Ignore this whole paragraph if it doesn't make sense! :smiley: )

lurch, 2017-07-03:
Another crazy thought - what about if instead of having a 'standalone' `RPiGPIOPin` class, it was actually just an internal-class of `RPiGPIOFactory` ? (again, ignore me if I'm talking gibberish)

waveform80, 2017-07-03:
> And what about "To prevent unpredictable results, should e.g. RPiGPIOPin's init check that factory is an instance of RPiGPIOFactory ?" ;-)

Currently there's no such check. I can throw one in, but I don't think we publicly document constructing pins via their class (anymore ... ahem :) so this shouldn't be something that end users ever attempt. Anyway, I'll bung that in a commit in a mo as it's only a few extra lines.

> Thanks for the explanation. I've not been python-ing for a while, but does it have anything like 'singleton objects' so that if RPiGPIOPin was passed a None factory, it'd get the singleton [...]

It does make sense :) However, currently the factories aren't singletons. The RPiGPIOFactory *could* be, but I figured if we didn't absolutely *have* to mess around in `__new__` we probably shouldn't just to try and keep the code reasonably simple (for the sake of future maintainers sanity!). But I can see there's a decent argument to be made that they should be singletons just to prevent people shooting themselves in the foot if they construct RPiGPIOFactory multiple times (even implicitly, given it's the default `Device.pin_factory`).

> Another crazy thought - what about if instead of having a 'standalone' RPiGPIOPin class, it was actually just an internal-class of RPiGPIOFactory ? (again, ignore me if I'm talking gibberish)

Again, definitely not gibberish - it's a perfectly valid idea. The pin implementations could indeed be private classes, but I couldn't remember when I wrote them (or now!) whether there's anything special about inheriting from private classes, so I didn't bother (e.g. RPiGPIOPin descends from LocalPin, which descends from PiPin, which descends from Pin so there's quite a bit of inheritance still going on).

Might be worth looking into though...

lurch, 2017-07-03:
Could just the 'concrete implementations' (RPIOPin, NativePin, etc.) be private classes, with the super-classes (which are basically 'interfaces') remaining public (LocalPin, PiPin, etc.) ?
But I'm sure this is something you'll investigate fully anyway ;-)

> for the sake of future maintainers sanity!

I'm pretty sure that some of this internal "reasonably simple" code is stuff that only @waveform80 fully understands anyway ;-)

commit 978f8175 (2017-07-14): Merge pull request #569 from waveform80/pin-factory-docs

Fix #568, fix #572, fix #565
commit eafae5c3 (2017-07-14): Fix #568

Update examples in pin factory docs so they work correctly.
closed 2017-07-14 by waveform80