gpiozero/gpiozero #571: How to initialize led?

makonik, 2017-07-09:
The documentation included in GPIOZero says use simply:
led=LED(number)
When I do it, my leds do not switch off properly. However, when I use 
led=LED(number, False) or led=LED(number, False, False)
leds are working correctly.
I am not sure now where did I find this definition and why the official definition does not work.
Is there something what I am missing?

bennuttall, 2017-07-09:
When you say "my leds do not switch off properly" do you mean when using on/off methods, or at the end of your script?

A few things you should know:

The constructor options for LED and their default values are as follows:

```python
LED(pin=None, active_high=True, initial_value=False)
```

And the docstring:

```
:param int pin:
    The GPIO pin which the LED is attached to. See :ref:`pin_numbering` for
    valid pin numbers.

:param bool active_high:
    If ``True`` (the default), the LED will operate normally with the
    circuit described above. If ``False`` you should wire the cathode to
    the GPIO pin, and the anode to a 3V3 pin (via a limiting resistor).

:param bool initial_value:
    If ``False`` (the default), the LED will be off initially.  If
    ``None``, the LED will be left in whatever state the pin is found in
    when configured for output (warning: this can be on).  If ``True``, the
    LED will be switched on initially.
```

LEDs can either be active high or active low. Active high means setting the pin high turns on the LED, and setting the pin low turns off the LED. This is the most common. Some LEDs have the opposite behaviour, which could be your problem. Simply setting `active_high` to `False` will mean `led.on()` and `led.off()` work as expected despite the difference in underlying electronics:

```python
from gpiozero import LED

led = LED(17, active_high=False)
led.on()
```

The `initial_value` parameter is to set the LED's initial state: on (True) or off (False):

```python
from gpiozero import LED

led = LED(17, initial_value=False)  # the LED came on immediately without using led.on()
```

Your example:

```python
led=LED(number, False, False)
```

is using positional arguments for `pin`, `active_high` and `initial_value`, just passed in order instead of named. So you are actually setting it to active high, but still off by default.

Another thing you should know is that a Python script with just:

```python
from gpiozero import LED

led = LED(17)
led.on()
```

will appear to do nothing. At the end of the script, the pin is reset to its original state, and there's nothing keeping the script running (like a while loop or [`signal.pause()`](http://gpiozero.readthedocs.io/en/stable/notes.html#keep-your-script-running). So the LED may come on for a brief moment but immediately turn off again.

I hope that clears it up. If you still have an issue, please paste your code.

makonik, 2017-07-10:
thank you for in depth explanation. Now, when I understand it better, I wil check my code and write again.

closed 2017-07-27 by bennuttall