gpiozero/gpiozero #572: failed to load factory

topshed, 2017-07-11:
pi@raspberrypi:~ $ git clone https://github.com/rpi-distro/python-gpiozero
Cloning into 'python-gpiozero'...
remote: Counting objects: 3669, done.
remote: Compressing objects: 100% (5/5), done.
remote: Total 3669 (delta 0), reused 2 (delta 0), pack-reused 3664
Receiving objects: 100% (3669/3669), 6.82 MiB | 349.00 KiB/s, done.
Resolving deltas: 100% (2450/2450), done.
Checking connectivity... done.

pi@raspberrypi:~ $ virtualenv -p python3 gp0-env
Already using interpreter /usr/bin/python3
Using base prefix '/usr'
New python executable in gp0-env/bin/python3
Also creating executable in gp0-env/bin/python
Installing setuptools, pip...done.

pi@raspberrypi:~ $ source gp0-env/bin/activate

(gp0-env)pi@raspberrypi:~ $ cd python-gpiozero/

(gp0-env)pi@raspberrypi:~/python-gpiozero $ python setup.py develop
running develop
running egg_info
creating gpiozero.egg-info
writing gpiozero.egg-info/PKG-INFO
writing requirements to gpiozero.egg-info/requires.txt
writing dependency_links to gpiozero.egg-info/dependency_links.txt
writing entry points to gpiozero.egg-info/entry_points.txt
writing top-level names to gpiozero.egg-info/top_level.txt
writing manifest file 'gpiozero.egg-info/SOURCES.txt'
reading manifest file 'gpiozero.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
writing manifest file 'gpiozero.egg-info/SOURCES.txt'
running build_ext
Creating /home/pi/gp0-env/lib/python3.4/site-packages/gpiozero.egg-link (link to .)
Adding gpiozero 1.3.2 to easy-install.pth file
Installing pinout script to /home/pi/gp0-env/bin

Installed /home/pi/python-gpiozero
Processing dependencies for gpiozero==1.3.2
Finished processing dependencies for gpiozero==1.3.2

(gp0-env)pi@raspberrypi:~/python-gpiozero $ pip install pigpio
Downloading/unpacking pigpio
  Downloading pigpio-1.35.tar.gz
  Running setup.py (path:/tmp/pip-build-n1_fsqv1/pigpio/setup.py) egg_info for package pigpio
    
Installing collected packages: pigpio
  Running setup.py install for pigpio
    
Successfully installed pigpio
Cleaning up...

(gp0-env)pi@raspberrypi:~/python-gpiozero $ python3
Python 3.4.2 (default, Oct 19 2014, 13:31:11) 
[GCC 4.9.1] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from gpiozero import LED
/home/pi/python-gpiozero/gpiozero/devices.py:505: PinFactoryFallback: Failed to load factory rpigpio: No module named 'RPi'
  'Failed to load factory %s: %s' % (name, str(e))))
/home/pi/python-gpiozero/gpiozero/devices.py:505: PinFactoryFallback: Failed to load factory rpio: No module named 'RPIO'
  'Failed to load factory %s: %s' % (name, str(e))))
>>>

bennuttall, 2017-07-11:
That's expected at present. What you're seeing is not an error, but a warning. If you proceed to create an LED object it will work fine, using pigpio as the pin factory. It's just that there's no message on *success* of finding a pin library.

What it's doing is:

1. Look for RPi.GPIO
  - Not found, show warning
2. Look for RPIO
  - Not found, show warning
3. Look for pigpio
  - Found, stop looking for pin libraries, use pigpio

If pigpio wasn't found it would proceed to step 4: try to use NativePin.

However, the issue you raise is perfectly valid. This is new behaviour since the last release, and although I understand why it happens, I fear users will assume this is a fatal error.

@waveform80 is there any chance we can lower the level of this warning, so that it doesn't show by default? I think it has potential to cause confusion.

waveform80, 2017-07-11:
Well, the whole point of the warning is for people who've forgotten to install RPi.GPIO in their virtualenv's. I can quite easily disable the warnings but that does rather defeat the purpose of having them in the first place (in other words, it'd be better to simply remove them).

The only people who'll see these warnings are people who are trying gpiozero in a venv (because the Raspbian package "recommends" RPi.GPIO so it'll usually be available unless someone's deliberately uninstalled it). It also won't happen if someone explicitly specifies the factory they want (e.g. `GPIOZERO_PIN_FACTORY=pigpio python ...`) because in that case there's no fallback.

I could make the warning message a bit more verbose, e.g. "failed to load %s, falling back to another implementation" or something like that if that'd help? But otherwise we may as well just chuck the warning.

bennuttall, 2017-07-11:
I think we should consider something like that. Since it only affects virtualenv users (or those system-installing with pip on raspbian lite), it's not a huge problem. But the wording definitely needs changing to point out it's not fatal.

topshed, 2017-07-11:
I think the warnings should probably stay, but just be a little clearer. It just looked like a fatal error to me so I never considered carrying on. When something says it 'failed', that implies that what you're trying to do hasn't worked.  A suggestion to install RPi.GPIO might help - that was the  step I had missed out.

commit 4009bf39 (2017-07-14): Fix #572

Tempted to add a FAQ entry about this anyway as it's come up multiple
times
commit 978f8175 (2017-07-14): Merge pull request #569 from waveform80/pin-factory-docs

Fix #568, fix #572, fix #565
closed 2017-07-14 by waveform80