gpiozero/gpiozero #573: Colour Sensor

bsimmo, 2017-07-12:
I just had a look to see if you had a supported colour sensor and notice there isn't one.
Are there any plans to ?

The only one I've used is the TCS34725 as seen on the Enviro pHAT.

dglaude, 2017-07-12:
As far as I know, there is no I2C support in gpiozero.
So many sensor, that one included seems to be outside of the current scope.

The good thing about TCS34725 on Enviro pHAT or standalone (https://shop.pimoroni.com/products/adafruit-rgb-color-sensor-with-ir-filter) is that @pimoroni made a Python library.

I am stuck with this DF Robot sensor (https://shop.pimoroni.com/products/gesture-sensor) for witch I did not find python code, only Arduino library.

bennuttall, 2017-07-13:
As @dglaude said, we don't currently support I2C but we intend to add support in the next release (after the one that's due this month). I'm not sure what the support will be but it will at the very least provide fundamental support for the protocol, so hopefully you will be able to access the colour sensor in some way. Watch out for updates and you can help us test it!

ukBaz, 2017-07-13:
Can I gently nod towards https://github.com/intel-iot-devkit/mraa again?
It has support for I2C for Raspberry Pi and could be used as a pin factory.

bsimmo, 2017-07-13:
Excellent, it wasn't actually for me as such, I was replying to a question on the RPI forum and thought I would have a look it gpiozero supported it. 
A colour sensor wasn't there, thought best to ask in case something is there, not thought of, or just not in the current stable documents :-)

Aside/
I only have and know about the Enviro pHAT using it (and yes they have a simple library and I see Adafruit does too that can be used.) because of our local (uk) Raspi. JAM and connecting it the The Internet of Things LoRA gateway, other than reading about it in the Mag.
No idea if there are any other interfaces used by other colour sensors that may have been supported.

Quite happily play and test things out :-D

alexsavio, 2019-12-01:
I have been working with the Enviro+ and the particle matter sensor kit. I was a bit sad I could not use gpiozero for the setup, and also a bit sad about the state of the code quality of the tools I had to use instead. I wonder what lower level system support is missing to make that happen in gpiozero, is it I2C support? What are the blockers for that to happen? Maybe I can do something about it.

bennuttall, 2019-12-01:
Yeah, we don't have any I2C support at the moment. If you want to have a go at taking some base I2C code and making it more gpiozero-like, that would be handy.