gpiozero/gpiozero #575: SPIHardwareInterface: unable to read/change mode

Aydan, 2017-07-14:
reading or setting clock_mode/polarity/phase fails with error "'SPIHardwareInterface' object has no attribute 'mode'"
functions _get_clock_polarity, _set_clock_polarity, _get_clock_phase, _set_clock_phase use self.mode instead of self._device.mode

waveform80, 2017-07-14:
Pretty sure this is already fixed for 1.4, and my apologies but the API for obtaining an SPI interface has changed quite a bit in this version, primarily for the purpose of supporting remote hardware SPI properly. That said, the actual SPI interface itself (clock_mode, clock_polarity, read, write, transfer, etc.) is all the same so hopefully it won't mean too much pain for anyone using it.

Aydan, 2017-07-14:
I've been using gpiozero.spi.SPI but I'll probably refactor to inherit from SPIDevice.
When will 1.4 hit the repos?

waveform80, 2017-07-14:
It's not released yet, but I'm working through the final bits now - shouldn't be more than a day or two now. If you're inheriting from `SPIDevice` you shouldn't have any changes to make (the SPI interface will be constructed for you by the super-class and you can use it exactly as you have been ... but without the bugs ;).

waveform80, 2017-07-14:
Anyway, I may as well mark this as for the 1.4 milestone as it deserves to be in the changelog, but I'll close it as I can't reproduce on the current master (with either local or remote SPI interfaces), and in 1.4 we've got tests for the various SPI interface attributes. Still, many thanks for the report - it's definitely useful!

closed 2017-07-14 by waveform80